lvgl/lv_platformio #1: Improve platformio demo/tutorial

puzrin, 2019-03-26:
In continuation of #667

## Goals

Simplify quick-start for beginers and promote modern best practices

1. Allow setup as simple as possible (1-click preferred).
2. Allow quick start with interface prototyping without hardware.
3. Provide good example of project structure.

## Suggestion

1. Create demo/template with platformio (improve [existing](https://github.com/littlevgl/pc_simulator_sdl_platformio))
   - When project opened in vscode, it will install as much as possible automatically for current OS (platformio itself, compilers, flashers, lib dependencies, ...).
2. Support most demanded pio platforms:
   - native - mandatory, for prototyping on PC.
   - stm32 - desired. [Example 1](https://ru.aliexpress.com/store/product/Free-shipping-STM32F429-DISCO-Embeded-ST-LINK-V2-STM32-Touch-Screen-Evaluation-Development-Board-STM32F4-Discovery/1534002_32228706345.html), [Example 2](https://ru.aliexpress.com/item/Free-shipping-STM32-STM32F103VCT6-HY-MINI-Development-Board-With-3-2-TFT-LCD-Module/451867272.html), [example 3](https://ru.aliexpress.com/item/STM32-Arduino-UNO-R3-USB/32859788497.html) + [display](https://ru.aliexpress.com/item/2-4-TFT-Arduino-UNO-R3-Mega2560/32859804381.html)
   - esp32 - optional. [Example 1](https://ru.aliexpress.com/item/TTGO-TS-v1-0-v1-4-esp32-1-44-1-8/32847643640.html), [example 2](https://ru.aliexpress.com/item/ESP32-Bluetooth-WI-FI-OLED-0-96-CP2102/32878668364.html), [example 3](https://ru.aliexpress.com/item/Wi-Fi-Bluetooth-esp32-4-flash-UNO-D1-R32-Arduino/32920335664.html) + display. 
3. Suggest good project structure, to minimize platform-specific code.

## For discuss

- Should demo be RTOS-based to minimize difference between platforms (delays, timers, internal communications)?
- What platforms should be supported?
  - native - mandatory
  - stm32/esp32 - both optional, just to prove code is working and suggested project structure is convenient.
- Is it possible to use CubeMX for stm32 specific code setup/update? IMHO it's convenient for beginners and has nice GUI. But it's horrible for integration, because contains `main()` in generated code.
- Display content?
  - transparent overlay for debug messages (convenient)
  - "settings" menu with +/- incremented params (popular demand)
  - animated graph (just for fun)
- anything else missed?

puzrin, 2019-04-16:
not stale

puzrin, 2019-05-08:
not stale

closed 2019-06-05 by stale[bot]

puzrin, 2019-08-02:
Does anyone plan to fix demo for `STM32F429-discovery` board? I can create issue with summary, but would like to avoid coding.

kisvegabor, 2019-08-02:
Please create an issue to summarize the current state.