lvgl/lv_platformio #2: Problem building

andrimarjonsson, 2019-07-26:
Been trying to get this going for the better part of a a day. Instructed in the forum to post here. Link to the thread can be found [here](https://forum.littlevgl.com/t/problem-building-sdl-demo-on-vscode-platformio/326).

When I try to build, this is the output I'm presented with:
```Processing native (platform: native)
Verbose mode can be enabled via -v, --verbose option
PACKAGES:
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 2 compatible libraries
Scanning dependencies…
Dependency Graph
|-- 6.0
|-- <littlevgl_sdl_drivers> 1.1
Compiling .pio/build/native/src/demo.o
Compiling .pio/build/native/src/img_bubble_pattern.o
Compiling .pio/build/native/src/main.o
Compiling .pio/build/native/lib72b/littlevgl_ID6228/lv_core/lv_disp.o
In file included from src/demo.c:9:
include/demo.h:18:10: fatal error: ‘lvgl.h’ file not found
#include <lvgl.h>
^~~~~~~~
1 error generated.
src/img_bubble_pattern.c:2:10: fatal error: ‘lvgl.h’ file not found
#include <lvgl.h>
^~~~~~~~
*** [.pio/build/native/src/demo.o] Error 1
…```
I’ve tried running “Update project libraries” to no avail. The dependent libs(lvgl and the sdldrivers) are located in .pio/libdeps/native so I have no idea why the PIO build isn't picking them up, thought PIO was made to exactly solve this problem, that is not to manually have to manage you include paths and libs?

puzrin, 2019-07-26:
This demo is ahead of time :), it just doesn't work and need rewrite.

puzrin, 2019-07-26:
Code from branch `vit` works, need PIO 4.+. It uses lvgl 5.3

But note, that's all internal experiments, not for public use. This repo needs full rework. Use on your own risk, and without support.

andrimarjonsson, 2019-07-28:
Fair enough. Thank you for a quick reply

Might I suggest then, a big disclaimer in the official documentation saying that not all environments are ready. As it stands now there isn't any mention of something being in beta or similar in the docs.

puzrin, 2019-07-31:
I suddenly managed to fix it for 6.+. Enjoy! So, created more issues where you can see details. IMO everything will be solved in nearest time.

Can we close this?

puzrin, 2019-08-02:
Closed - should work now. See pending issues for the rest questions and feel free to create new ones.

closed 2019-08-02 by puzrin