lvgl/lv_platformio #3: Pre-release pending tails

puzrin, 2019-07-31:
This things should be done to announce PIO demo as "useable":

- [ ] fix hardcoded paths in `lv_drivers` https://github.com/littlevgl/lv_drivers/issues/66#issuecomment-516810456
- [ ] add pio config (check first), tag new version and publish `lv_drivers`
- [ ] publish `lvgl` with new tag (because correct pio config was added)
- [ ] Drop outdated branches
- [ ] Drop https://github.com/littlevgl/platformio_sdl_drivers repo
- [ ] Rename this repo.

puzrin, 2019-07-31:
@kisvegabor, @embeddedt see checklist, need your care.

kisvegabor, 2019-08-01:
> fix hardcoded paths in lv_drivers littlevgl/lv_drivers#66 (comment)

Done

> lv_drivers: add pio config (check first), tag new version and publish

Please send a PR

> publish lvgl with new tag (because correct pio config was added)

Probably I'll today

> Consider rename this repo, #4.

I've commented on it

puzrin, 2019-08-01:
I've updated demo. It's almost useable :).

Last step - put new tags for lvgl & lv_drivers, and i can switch dependencies from master branches to normal numbers.

puzrin, 2019-08-01:
@kisvegabor i see you tagged `lvgl`. Could you add new tag to `lv_drivers` too. Need at least 1 tag with correct PIO config.

embeddedt, 2019-08-01:
https://github.com/littlevgl/lv_drivers/releases/tag/v6.0.1

puzrin, 2019-08-02:
https://platformio.org/lib/show/6598/lvgl

Hm... i don't see tags. May be PIO wish numerics?

kisvegabor, 2019-08-02:
The [documentation](https://docs.platformio.org/en/latest/librarymanager/config.html?highlight=library.json#version) says it `v` prefix is ok.

Probably it needs the usual 24 hours to be updated.

puzrin, 2019-08-02:
According to description, seems we should add `version` field with numeric (without `v`) value to force crawler scan git tags.

I don't know why they did so, but description looks like this.

puzrin, 2019-08-02:
https://community.platformio.org/t/missed-versions-git-tag-names-in-published-package/9005 - asked on PIO forum.

PS. but with high probability we will have to add versions & reloase 6.0.2 for lvgl & lv_drivers.

puzrin, 2019-08-02:
https://community.platformio.org/t/missed-versions-git-tag-names-in-published-package/9005/2?u=vit

Could you add `"version": "6.0.2",` to `lvgl` & `lv_drivers` configs and release 6.0.2?

kisvegabor, 2019-08-02:
Uh, that's sad... I wanted to keep the monthly minor releases. The conclusion is we need to test more before even patch releases to avoid releases every two days.

I suggest trying it with `lv_driver` first. Could you add the required changes?

puzrin, 2019-08-02:
> The conclusion is we need to test more before even patch releases to avoid releases every two days.

Sorry :(, sometime it's technically impossible to test everything and shit happens. I had no experience with publishing "proper" PIO libs before (`libfixmath` has ho tags at all).

> I suggest trying it with `lv_driver` first. Could you add the required changes?

Good idea. Proposed PR, but i have no permissions to set git tags.

kisvegabor, 2019-08-02:
> Sorry :(, sometime it's technically impossible to test everything and shit happens. I had no experience with publishing "proper" PIO libs before (libfixmath has ho tags at all).

No problem, just a little bit pains inside  :smile: 

> Good idea. Proposed PR, but i have no permissions to set git tags.

Done

puzrin, 2019-08-06:
Seems done. Pushed update.

Last thing is switch `lvgl` dependency from master to numeric (as with drivers).

closed 2019-08-06 by puzrin

puzrin, 2019-08-07:
Updated `lvgl` dependency to numeric. Now everything seems useable.

I think it's possible to simplify SDL install for windows (via pio package with pre-built binaries & headers). But i use linux only, and here everything is fine.

kisvegabor, 2019-08-07:
> Updated lvgl dependency to numeric. Now everything seems useable.

Great, thank you for the feedback!