lvgl/lv_platformio #4: Repo rename

puzrin, 2019-07-31:
Current name does not reflects repo goals. It's not about sdl but about how to make everything simple & convenient. So, `pc_simulator_sdl_` looks like irrelevant buzzword.

1. I suggest to rename into `platformio_lvgl_demo`. Or something else of suck kind.
2. See rewritten README in https://github.com/littlevgl/pc_simulator_sdl_platformio/tree/readme and let me know if i can merge it

kisvegabor, 2019-08-01:
> I suggest to rename into platformio_lvgl_demo. Or something else of suck kind

If this repo will be generic not only a simulator then we really should rename it.
For consistency, I'd like `lv_platformio`. (This convention is already used for some repos but not all of them yet.)

puzrin, 2019-08-01:
> For consistency, I'd like `lv_platformio`.

I like it. Better than my one.

What about new README in branch?

kisvegabor, 2019-08-01:
> I like it. Better than my one.

Renamed

> What about new README in branch?

I like it. Simple and easy to follow.

closed 2019-08-01 by puzrin