lvgl/lv_platformio #6: Add STM32F429-discovery support

puzrin, 2019-08-02:
Summary from #1.

It would be nice add `STM32F429-discovery` support or remove build target to avoid users confusion. Adding support is preferable.

## What we have now

1. Skeleton to easy extend demo with new targets and keep common files free of cryptic guards:
   - https://github.com/littlevgl/lv_platformio/tree/master/drivers
   - https://github.com/littlevgl/lv_platformio/blob/master/platformio.ini
2. Discussion how to activate LCD drivers (were build errors):
   - https://community.platformio.org/t/how-in-include-disco-f429zi-headers/7001/13?u=vit (see also up and down)
   - ignore mentions of `variant.json` edit - already fixed.
3. Something outdated in stale branch https://github.com/littlevgl/lv_platformio/tree/stm - i don't know if it has something reuseable, so not deleted yet.

## What would be nice to get

Minimal case:

- HAL-based code, runnable on `STM32F429-discovery`, with display & touch support.

Maxmal case:

- Previous + third target, `mbed`-based (instead of HAL). Reasons - people like to use `mbed` and create many drivers for it. 

---

Notes:

- Avoid use `lv_drivers` (except SDL case). PIO has huge number of driver packages for different peripherals & boards. Need only small local wrappers for lvgl api.
- https://github.com/platformio/platform-ststm32/tree/master/examples - may be useful to inspect.

kisvegabor, 2019-08-02:
So for the first try would it work to add the existing tft and touchpad driver (with the required updates) to `stm32f429_disco` folder and initialize them in `driver.c` in `hw_init()`?

puzrin, 2019-08-02:
~ Yes. `driver.c` should also have all mandatory interrupt handlers (if HAL not provide those, i don't know details).

Also, not sure is it worth call anything  from hw_loop(). May be worth use HAL's 1ms system ticks or timer. There are no any restriction HOW to implement details. The only idea is to isolate platform-dependent code.

kisvegabor, 2019-08-02:
I'll give it a try next week. If somebody wants to try it before that, go ahead!

puzrin, 2019-08-23:
IMO those frequent stalebot messages are annoying. At least, i see not reason use that bot in this repo.

kisvegabor, 2019-08-24:
I've created the `pinned` label. Pinned issues won't be marked as stale.

puzrin, 2019-08-24:
I don't know why it may worth mark all as pinned instead of just disabling stalebot. I'd suggest enable it only after there appears 2-3 issues where it could be useful.

embeddedt, 2019-08-24:
> I don't know why it may worth mark all as pinned instead of just disabling stalebot.

Because, in many cases, people open issues to report a bug/ask a question, we fix the bug/respond, and they do not write anything back or close the issue.

Stale saves us from having to check all the repositories every month and close the old issues.

puzrin, 2019-08-24:
> Because, in many cases, people open issues to report a bug/ask a question, we fix the bug/respond, and they do not write anything back or close the issue.

That's good in theory, but i speak about particular case. Here are no users, developers only. As a result - bot kicks developers instead of intended target.

kisvegabor, 2019-08-26:
Thre are really not too much issue here. I've disabled `stale` here.

puzrin, 2019-10-05:
Let's close it, since reached acceptable state.

closed 2019-10-05 by puzrin