lvgl/lv_platformio #7: <SDL2/SDL.h> and other header files missing

Chronic788, 2019-09-03:
When I cloned the source code and followed the instructions to build using VSCode (found here: https://github.com/littlevgl/lv_platformio) I found that these header files are missing:

- <SDL2/SDL.h>
- indev/mouse.h
- indev/mousewheel.h
- indev/keyboard.h

embeddedt, 2019-09-03:
Are you on Windows or Linux?

Chronic788, 2019-09-03:
I am on Windows 64 bit. I've followed the instructions to install SDL2 from this page: https://docs.littlevgl.com/en/html/get-started/pc-simulator.html#install-sdl-2

After looking at things further, it seems my installation of MinGW is a little different than what the above link. I went to this page: http://mingw-w64.org/doku.php/download
and downloaded the MingW-W64-builds selection from sourceforge. It installed MinGW to C:\Program Files (x86) which is different than the documentation instructions say: they say it should be:  C:/MinGW/…/x86_64-w64-mingw32. Then I followed the instructions to copy the files from the download of the development files found here: https://www.libsdl.org/download-2.0.php . I am not sure that my MinGQ install is the problem, though. Would the copying of the SDL2 files be different for VSCode than Eclipse?

puzrin, 2019-09-03:
I use linux, no ideas how to install sdl2 for win.

About `indev` - i had a very strange bug with different paths behaviour in windows/linux https://github.com/speedcontrols/ac_sc_grinder/pull/2, but not sure if it applies to this case. It really worth fix sdl2 path first.

embeddedt, 2019-09-04:
@Chronic788 On Windows it's usually easiest to use the Visual Studio or CodeBlocks projects instead. They don't depend on an external library like SDL and thus are easier to set up.

Chronic788, 2019-09-04:
I've resolved my problem by switching to Linux

closed 2019-09-04 by Chronic788