lvgl/lv_platformio #8: Build fails due to missing lvgl.h

jsphpl, 2019-09-05:
When i run `platformio build`, i get the following error:

```
Processing stm32f429_disco (platform: ststm32; board: disco_f429zi; framework: stm32cube)
------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Warning! Library `{'requirements': '^6.0', 'name': 'LittlevGL'}` has not been found in PlatformIO Registry.
You can ignore this message, if `{'requirements': '^6.0', 'name': 'LittlevGL'}` is a built-in library (included in framework, SDK). E.g., SPI, Wire, etc.
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/disco_f429zi.html
PLATFORM: ST STM32 > ST 32F429IDISCOVERY
HARDWARE: STM32F429ZIT6 180MHz 256KB RAM (2MB Flash)
DEBUG: CURRENT(stlink) ON-BOARD(stlink) EXTERNAL(blackmagic, jlink)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 31 compatible libraries
Scanning dependencies...
No dependencies
Compiling .pioenvs/stm32f429_disco/drivers/stm32f429_disco/driver.o
Compiling .pioenvs/stm32f429_disco/drivers/stm32f429_disco/stm32f4xx_it.o
Compiling .pioenvs/stm32f429_disco/drivers/stm32f429_disco/system_stm32f4xx.o
Compiling .pioenvs/stm32f429_disco/src/demo.o
In file included from src/demo.c:9:0:
src/demo.h:18:10: fatal error: lvgl.h: No such file or directory

**************************************************************
* Looking for lvgl.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:lvgl.h"
* Web  > https://platformio.org/lib/search?query=header:lvgl.h
*
**************************************************************

#include <lvgl.h>
^~~~~~~~
compilation terminated.
*** [.pioenvs/stm32f429_disco/src/demo.o] Error 1
drivers/stm32f429_disco/driver.c:3:10: fatal error: stm32f429_disco/stm32f429i_discovery.h: No such file or directory
#include "stm32f429_disco/stm32f429i_discovery.h"
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
drivers/stm32f429_disco/stm32f4xx_it.c:21:10: fatal error: lvgl/lv_hal/lv_hal_tick.h: No such file or directory
#include "lvgl/lv_hal/lv_hal_tick.h"
^~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/stm32f429_disco/drivers/stm32f429_disco/driver.o] Error 1
*** [.pioenvs/stm32f429_disco/drivers/stm32f429_disco/stm32f4xx_it.o] Error 1
drivers/stm32f429_disco/system_stm32f4xx.c:139:0: error: unterminated #if
#if BUILD_ENV_NAME == stm32f429_disco

drivers/stm32f429_disco/system_stm32f4xx.c:132:0: error: unterminated #if
#if BUILD_ENV_NAME == stm32f429_disco

drivers/stm32f429_disco/system_stm32f4xx.c:66:0: error: unterminated #if
#if BUILD_ENV_NAME == stm32f429_disco

*** [.pioenvs/stm32f429_disco/drivers/stm32f429_disco/system_stm32f4xx.o] Error 1
============================================================================ [ERROR] Took 1.86 seconds ============================================================================

==================================================================================== [SUMMARY] ====================================================================================
Environment native         	[ERROR]
Environment stm32f429_disco	[ERROR]
============================================================================ [ERROR] Took 3.34 seconds ============================================================================
```

Trying to install lvgl using `platformio lib install lvgl` yields the following output:

```
************************************************************************************************************************************************************************************
Obsolete PIO Core v3.6.7 is used (previous was 4.0.3)
Please remove multiple PIO Cores from a system:
https://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
************************************************************************************************************************************************************************************
Library Storage: /Users/jsphpl/clients/cheese/lv_platformio/.piolibdeps
LibraryManager: Installing id=6598
lvgl @ 6.0.2 is already installed
```

Can someone please point me into the right direction?

Thanks a lot!

puzrin, 2019-09-05:
First, `stm32f429_disco` target does NOT work and does not have proper "drivers". As far as i understand default build should assemble only `native` target if you do not set name explicit.

If someone decide to improve stm32 support, start point would be to update [this](https://github.com/littlevgl/lv_platformio/blob/master/platformio.ini#L46) line according to `native` target config and follow #6.

Consider current repo as "proof of concept", not as tutorial. Current progress stopped at point "lvgl is packed for PIO & sdl2 can be used".

jsphpl, 2019-09-05:
Alright, thanks for pointing that out, @puzrin !

closed 2019-09-05 by jsphpl