lvgl/lv_platformio #11: problem building: "register 'sp' unsuitable for global register variables on this target" (and other errors)

paulpv, 2019-10-05:
Machine: MacOS Mojave 10.14.6 Visual Studio Code Version: 1.38.1

Build output:
```
> Executing task: platformio run <

Processing native (platform: native)
--------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
PACKAGES: 
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 2 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <lvgl> 6.0.2
|-- <lv_drivers> 6.0.2
|   |-- <lvgl> 6.0.2
Compiling .pio/build/native/src/syscalls.o
Compiling .pio/build/native/lib4f4/lvgl_ID6598/lv_draw/lv_draw.o
Compiling .pio/build/native/lib4f4/lvgl_ID6598/lv_draw/lv_draw_arc.o
Compiling .pio/build/native/lib4f4/lvgl_ID6598/lv_draw/lv_draw_basic.o
src/syscalls.c:65:31: error: register 'sp' unsuitable for global register variables on this target
register char * stack_ptr asm("sp");
                              ^
src/syscalls.c:116:1: error: unknown type name 'caddr_t'
caddr_t _sbrk(int incr)
^
src/syscalls.c:131:11: error: use of undeclared identifier 'caddr_t'
                return (caddr_t) -1;
                        ^
src/syscalls.c:136:10: error: use of undeclared identifier 'caddr_t'
        return (caddr_t) prev_heap_end;
                ^
4 errors generated.
*** [.pio/build/native/src/syscalls.o] Error 1
============================= [FAILED] Took 4.31 seconds =============================

Environment      Status    Duration
---------------  --------  ------------
native           FAILED    00:00:04.307
stm32f429_disco  IGNORED
======================= 1 failed, 0 succeeded in 00:00:04.307 =======================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.
```

¯\_(ツ)_/¯

puzrin, 2019-10-05:
PIO support is very fresh and consists only of publishing some packages with correct manifests.

This demo was tested only with linux. I guess, all improvements are possible via PRs only :(. As far as i understand, you should be able to do stm32 build.

puzrin, 2019-10-05:
I reworked notes, to exclude any possible confusion about non-linux builds of `native` target.

Closing as `wontfix`. If anyone has knowledge and time to make `native` demo work in Windows/macOS - PR welcome. Current state is NOT pio or lvgl limitation. It's related to this demo only. This demo was done by volunteers as good as they were able :)

As alternative - try [other IDEs](https://docs.littlevgl.com/en/html/get-started/pc-simulator.html#select-an-ide).

closed 2019-10-05 by puzrin