lvgl/lv_platformio #12: lvgl (core) for PlatformIO IDE fails to compile when installed - Several Issues to fix it

jmulchin, 2019-11-12:
> # Important: issues that don't use this template will be ignored/closed.

**Describe the bug**
Installing lvgl (core) using the PlatformIO IDE fails to compile without errors. Failures are related to header files and directory references included during compile process.

A clear and concise description of what the bug is.
Problem littlevgl/lvgl#1 - The installer places the library at C:\users\[username]\.platformio\lib\lvgl_ID6598
That's Ok, but the root directory is no longer called lvgl, it is now called lvgl_ID6598. So any references in header files to lvgl/src/something is broken. Sorry, don't remember the files(s) I needed to fix this problem, but I think the fix for this was to remove the first level "lvgl/" from the include statement.
Problem littlevgl/lvgl#2. - Most of the Header Files have an #include "../../../lv_conf.h" statement. This causes the compiler to look one level too high in the path. The fix is to change all header files to say
#include "../../lv_conf.h, removing one level of directory search. There are a lot of files to change.

**To Reproduce**
Do a clean Install under PlatformIO IDE and attempt to compile the library. You need to add the lib_deps = lvgl to the platformio.ini file to get it to see the new library.

Let me know if you need any other information or help with this problem.

Jerry

kisvegabor, 2019-11-12:
I've moved this issue to the PlatformIO repo.

cc @puzrin

puzrin, 2019-11-12:
I'm not sure this issue is related to this repo. Looks like misconfigured project. I'd suggest first to clone this repo and try stm32 buid (emulator will not work)

puzrin, 2019-11-15:
Close - no feedback, looks like project misconfiguration.

closed 2019-11-15 by puzrin

nmathon, 2019-11-17:
Hi,

I have exactly the same problem when trying to integrate the lvgl library in platformio.
Who is the owner of this library ? Is it related to this project?

The compilation fail with :
In file included from drivers/driver.c:4:0:
drivers/tft/tft.h:13:10: fatal error: lvgl/src/lv_misc/lv_color.h: No such file or directory

And that's true the lvgl library is referenced under lvgl_ID6598, which shouldn't.

puzrin, 2019-11-17:
`lv_platformio` was done mostly by me, as an example, and not actively maintained. It's buildable under Ubuntu. stm32 target should be buildable under windows, but nobody tested. All pio "support" consists of configs in `lvgl` & `lv_drivers` repos.

Any improvements are welcome.

In my [real project](https://github.com/puzrin/dispenser) i use more fresh codebase with better stm32cube integration, but you sould use Cube 5.0.+ to generate compatible hal version (5.1.+ will not work).

Could you describe exactly step-by-step your actions, starting from repo clone?