lvgl/lv_platformio #13: Macro errors when compiling fresh out of the box

wthie, 2020-01-04:
On Ubuntu 19 not touching anything after git clone  and running pio for native target in either terminal or vscode I get the following errors:

In file included from .pio/libdeps/native/lvgl_ID6598/lvgl.h:19,
                 from src/demo.h:18,
                 from src/demo.c:9:
.pio/libdeps/native/lvgl_ID6598/src/lv_misc/lv_log.h:29:28: error: expected declaration specifiers or ‘...’ before numeric constant
   29 | #define LV_LOG_LEVEL_TRACE 0 /**< A lot of logs to give detailed information*/
      :
      :
ad lib

----- Version info --------------------------------------------------------------------------
Distributor ID:	Ubuntu
Description:	Ubuntu 19.10
Release:	19.10
Codename:	eoan

gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/9/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:hsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 9.2.1-9ubuntu2' --with-bugurl=file:///usr/share/doc/gcc-9/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,gm2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-9 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --with-target-system-zlib=auto --enable-multiarch --disable-werror --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none,hsa --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu
Thread model: posix

closed 2020-01-04 by puzrin

puzrin, 2020-01-04:
Thanks. That happened because demo was done for lvgl 6.0 with too weak versions dependencies, and 6.1 added one more macro to config.

Should work now.

wthie, 2020-01-04:
Mahalo nui loa!

commit b48bd39b (2020-01-04): Fix for 6.1 + pio cfg cleanup, close #13