lvgl/lv_platformio #17: Add Windows support

puzrin, 2020-03-12:
Demo now supports Linux & Macos. Windows support missed. Unfortunately, i don't use Windows and have no time to add this feature. But i can help with PIO scripting and CI tests if anyone wish to care about Win.

Suggested plan is:

- Clone this repo for standalone experiments, to not care about other OS-es support.
- Create repo with SDL2 dlls & headers, to be used as dependency lib.
- Add created repo as library to `platform.ini` (don't worry about other OS-es at this step)
- Try to build & add search paths to build options if needed
- If works - kick me, to help with PR prepare.

I also prepared [template](https://github.com/littlevgl/lv_platformio/blob/master/.travis.yml#L41-L58) for CI testing.

timmcarson, 2020-04-05:
Hi @puzrin 

I would like to help with this.  
I am using:
  Windows 10
  VS Code
  PIO

I cloned the git

$ git clone https://github.com/littlevgl/lv_platformio.git
Cloning into 'lv_platformio'...
remote: Enumerating objects: 70, done.
remote: Counting objects: 100% (70/70), done.
remote: Compressing objects: 100% (49/49), done.
remote: Total 233 (delta 26), reused 58 (delta 17), pack-reused 163
Receiving objects: 100% (233/233), 659.40 KiB | 2.38 MiB/s, done.
Resolving deltas: 100% (102/102), done.

Compile Error attached
[error.log](https://github.com/littlevgl/lv_platformio/files/4432765/error.log)

puzrin, 2020-04-05:
You can just comment [appropriate](https://github.com/littlevgl/lv_platformio/blob/master/platformio.ini#L33) string in config.

timmcarson, 2020-04-05:
Sorry, I don't know how to comment the appropriate string

puzrin, 2020-04-06:
Then i think it's better to wait until someone suggests working soluion. As i said in first post, i have no windows and no plans to spend time about it.

embeddedt, 2020-04-06:
@timmcarson Adding a semicolon before `!python` should be enough.

timmcarson, 2020-04-06:
@embeddedt Thanks,

I tried that and then I got a gcc error
[error2.log](https://github.com/littlevgl/lv_platformio/files/4438682/error2.log)

embeddedt, 2020-04-06:
It seems that you don't have a compiler installed, or the wrong one is being chosen.

timmcarson, 2020-04-06:
That was my understanding.   am I right in thinking that the example should install everything needed or do I need to manually install?  

These are the extensions that I have installed currently
[image]

embeddedt, 2020-04-07:
I'm not sure how PlatformIO works, but with other IDEs on Windows, you're typically given an option to download the IDE+compiler or download the compiler separately. Because PlatformIO is an addon to Visual Studio Code, I'm not sure that the same options are available. @puzrin probably knows more about this than I do.

timmcarson, 2020-04-08:
Thanks everyone for the help.  

I changed gears and tried for a simple example.  I have been able to successfully run the ESP32_TFT_eSPI.ino Arduino example on an ESP32 using platformIO on a windows machine.

This is good enough for me.

thanks again.

puzrin, 2020-04-08:
@embeddedt https://travis-ci.org/github/littlevgl/lv_platformio/builds/672323062 bare metal builds work on all OS-es. Questions about sdl2 win builds are not expected IMO.

nickrodie, 2020-07-03:
To build emulator in Windows:

install MSYS2 : https://www.msys2.org/
install gcc : pacman -S mingw-w64-x86_64-gcc
install SDL2 : pacman -S mingw-w64-i686-SDL2

install VS Code : https://code.visualstudio.com/
install platformio : Extensions > search platformio
open lv_platformio folder
comment out the `!python -c 'import platform; print("-m32" if platform.system() in [ "Linux" ] else "")'` line in platformio.ini
build :)

kisvegabor, 2020-07-03:
@nickrodie That's great!

Can wee add this description to the README?

nickrodie, 2020-07-03:
Sure.

nickrodie, 2020-07-03:
@kisvegabor Should probably point out that those versions are quite specific. Namely minggw for 64bit Windows and SDL2. You might want to add that you can search for different vesrions in MSYS2 eg `pacman -Ss SDL`

kisvegabor, 2020-07-03:
I think having at least a working version for Windows is already great. 
@puzrin ?

puzrin, 2020-07-03:
@kisvegabor 

IMO this can be done more convenient:

1. Publish win sdl2 binaries in separate package
2. Add search path via -L flag.

In this case it will "just work". If anyone provides this kind of working example, i will help to do this compatible with anoter OS-es.

puzrin, 2020-07-03:
@nickrodie 

> comment out the `!python -c 'import platform; print("-m32" if platform.system() in [ "Linux" ] else "")'` line in platformio.ini

Why comment out needed? This line should add `-m32` only for linux, and nothing for win.

nickrodie, 2020-07-03:
> @nickrodie
> 
> > comment out the `!python -c 'import platform; print("-m32" if platform.system() in [ "Linux" ] else "")'` line in platformio.ini
> 
> Why comment out needed? This line should add `-m32` only for linux, and nothing for win.

It fails for me. `File "<string>", line 1
    'import
          ^
SyntaxError: EOL while scanning string literal
OSError: 'python -c 'import platform; print("-m32" if platform.system() in [ "Linux" ] else "")'' exited 1:  
  File "C:\Users\Nick\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 168:
    env.SConscript("$BUILD_SCRIPT")`
I can't give you much more information about it because I don't understand the whole build system well enough. Python is installed.

puzrin, 2020-07-03:
Thanks for info. Try to swap `"` and `'`. Like here: https://github.com/speedcontrols/ac_sc_grinder/blob/master/platformio.ini#L83. That code works on [all](https://github.com/speedcontrols/ac_sc_grinder/blob/master/.travis.yml) OS-es.

nickrodie, 2020-07-03:
IIRC I did actually manage to build SDL2 from https://www.libsdl.org/release/SDL2-devel-2.0.12-mingw.tar.gz but no matter what I tried I couldn't get lvgl to find all the files. I was following these instructions https://w3.cs.jmu.edu/bernstdh/Web/common/help/cpp_mingw-sdl-setup.php

nickrodie, 2020-07-03:
> Thanks for info. Try to swap `"` and `'`. Like here: https://github.com/speedcontrols/ac_sc_grinder/blob/master/platformio.ini#L83. That code works on [all](https://github.com/speedcontrols/ac_sc_grinder/blob/master/.travis.yml) OS-es.

Yep that fixed it `!python -c "import platform; print('-m32' if platform.system() in [ 'Linux' ] else '')"` 
`Environment      Status    Duration
---------------  --------  ------------
native           SUCCESS   00:00:04.455
stm32f429_disco  IGNORED`

puzrin, 2020-07-03:
Great! Let's fight with SDL2 to make it work in 1 click :).

Try this in your fork:

1. Create folder `/lib/lv_sdl2_win_bin`, and put there binaries. This will emulate extern package.
2. Try to make it work, with `-Llib/lv_sdl2_win_bin`. If needed - change name in `-lSDL2`.

Tips:

- Use verbose build, to check real generated options
- If lib name is non conventional - read this and below https://community.platformio.org/t/failing-to-link-external-library/14544/2, about `-l:` alternative

---

I'm not familiar with win toolchains. Could you explain why you install mingw version of gcc? AFAIK pio should provide something built-in for native platform. No?

nickrodie, 2020-07-03:
I don't think it will be as simple as 1 click for most.
gcc need to be installed for a start.
Here is output from a fresh clone into VS Code which already had platformio extension installed.
`> Executing task: C:\Users\Nick\.platformio\penv\Scripts\platformio.exe run <

Processing native (platform: native)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 2 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <lvgl> 6.1.1
|-- <lv_drivers> 6.1.1
|   |-- <lvgl> 6.1.1
Building in release mode
Compiling .pio\build\native\lib7e6\lvgl_ID6598\lv_core\lv_debug.o
'gcc' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\native\lib7e6\lvgl_ID6598\lv_core\lv_disp.o
'gcc' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\native\lib7e6\lvgl_ID6598\lv_core\lv_group.o
'gcc' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\native\lib7e6\lvgl_ID6598\lv_core\lv_indev.o
'gcc' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\native\lib7e6\lvgl_ID6598\lv_core\lv_obj.o
'gcc' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\native\lib7e6\lvgl_ID6598\lv_core\lv_refr.o
Compiling .pio\build\native\lib7e6\lvgl_ID6598\lv_core\lv_style.o
'gcc' is not recognized as an internal or external command,
operable program or batch file.
'gcc' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\native\lib7e6\lvgl_ID6598\lv_draw\lv_draw.o
Compiling .pio\build\native\lib7e6\lvgl_ID6598\lv_draw\lv_draw_arc.o
'gcc' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\native\lib7e6\lvgl_ID6598\lv_draw\lv_draw_basic.o
'gcc' is not recognized as an internal or external command,
operable program or batch file.
Compiling .pio\build\native\lib7e6\lvgl_ID6598\lv_draw\lv_draw_img.o
'gcc' is not recognized as an internal or external command,
operable program or batch file.
'gcc' is not recognized as an internal or external command,
Compiling .pio\build\native\lib7e6\lvgl_ID6598\lv_draw\lv_draw_label.o
operable program or batch file.
*** [.pio\build\native\lib7e6\lvgl_ID6598\lv_core\lv_debug.o] Error 1
*** [.pio\build\native\lib7e6\lvgl_ID6598\lv_core\lv_disp.o] Error 1
*** [.pio\build\native\lib7e6\lvgl_ID6598\lv_core\lv_group.o] Error 1
*** [.pio\build\native\lib7e6\lvgl_ID6598\lv_core\lv_indev.o] Error 1
*** [.pio\build\native\lib7e6\lvgl_ID6598\lv_core\lv_obj.o] Error 1
*** [.pio\build\native\lib7e6\lvgl_ID6598\lv_core\lv_refr.o] Error 1
*** [.pio\build\native\lib7e6\lvgl_ID6598\lv_core\lv_style.o] Error 1
*** [.pio\build\native\lib7e6\lvgl_ID6598\lv_draw\lv_draw.o] Error 1
*** [.pio\build\native\lib7e6\lvgl_ID6598\lv_draw\lv_draw_arc.o] Error 1
*** [.pio\build\native\lib7e6\lvgl_ID6598\lv_draw\lv_draw_basic.o] Error 1
*** [.pio\build\native\lib7e6\lvgl_ID6598\lv_draw\lv_draw_img.o] Error 1
'gcc' is not recognized as an internal or external command,
operable program or batch file.
*** [.pio\build\native\lib7e6\lvgl_ID6598\lv_draw\lv_draw_label.o] Error 1`
I will update as I get to SDL

nickrodie, 2020-07-03:
...
> I'm not familiar with win toolchains. Could you explain why you install mingw version of gcc? AFAIK pio should provide something built-in for native platform. No?
I don't know enough to answer that.

Also forgot to add in my original instructions that I had to add the `msys64\mingw64\bin` dir to path in windows.

What I can tell you is that as far as a Windows user experience goes the MSYS2 method is very simple. Just a few simple clicks and it's all done for you.

puzrin, 2020-07-03:
Hm... if user has to install gcc, then your choice of MSYS2 is reasonable, and it's not a problem to install stl2 from the same source. Then we can use your recipe as is. Not 1-click, but useable and simple enouth.

---

I don't know why but Travis docs suggest `choco` to install `python` https://github.com/puzrin/reflow_micro/blob/master/.travis.yml#L52. Is it built-in of Windows? Can it be used to simplify install of `MSYS2` https://chocolatey.org/search?q=gcc?

PS. I don't understand anything in this for win, only copy-pasted recipe.

nickrodie, 2020-07-03:
I think python is autpmatically installed as part of the platformio extension in vs code

nickrodie, 2020-07-03:
Also as far as installing gcc goes the msys method is also the easiest I tried. I did originally use mingw from http://www.mingw.org/wiki/Getting_Started but it took me a couple of goes to get all the right packages so that i could build SDL

nickrodie, 2020-07-03:
@kisvegabor @puzrin FYI I had a typo in original post and wrong SDL2 installation. I have amended the instructions.

puzrin, 2020-07-03:
> add path to mingw64\bin dir. typically C:\msys64\mingw64\bin

This is not convenient (need to think instead of copy-paste commands). Is there any way to autodetect location or add it to path automatically?

Did you tried `-LC:\msys64\mingw64\bin` instead?

nickrodie, 2020-07-03:
I don't think you'll have any luck making it any easier than what I've already indicated. Even MS talk about manually editing the path https://code.visualstudio.com/docs/cpp/config-mingw

embeddedt, 2020-07-03:
To make it easier I think you'd have to find an installer that has an "Add to PATH" option built in.

nickrodie, 2020-07-03:
However the path can be taken from the install if you use msys2 install dir + /mingw64/bin
[image]

puzrin, 2020-07-03:
>  don't think you'll have any luck making it any easier than what I've already indicated. Even MS talk about manually editing the path [code.visualstudio.com/docs/cpp/config-mingw](https://code.visualstudio.com/docs/cpp/config-mingw)

Sorry, i misread your post. I thought it's about path to sdl2. Then link to https://code.visualstudio.com/docs/cpp/config-mingw#_prerequisites should be ok.

PS. Last time i used windows, when path was added via editing `autorun.bat` file :)

nickrodie, 2020-07-03:
I mean you can add it with `setx /M PATH %PATH%;[install_dir]"` but this can fail under certain circumstances (like path too long) so it's not reliable.

puzrin, 2020-07-03:
This will be useful for Travis-CI anyway, i need add build test.

puzrin, 2020-07-03:
- Fixed win compatibility https://github.com/lvgl/lv_platformio/commit/38dd9ce985088338157510c8eb13c8fbe2661497
- Added `native` win build check to CI https://github.com/lvgl/lv_platformio/commit/04a2f24f2def168c03efd4e287a5a915489da68e.

@nickrodie Do you wish make PR to reflect your help in commits history?

nickrodie, 2020-07-03:
No. But thanks for the offer.

puzrin, 2020-07-03:
https://github.com/lvgl/lv_platformio/commit/39cb13e2da4f5711465efebb318d13fb95737297

Please, check. Any fixes needed?

puzrin, 2020-07-03:
Close as complete. Big thanks to @nickrodie for help!

closed 2020-07-03 by puzrin