lvgl/lv_platformio #18: Still issues with finding the lv_conf.h

DavidTruyens, 2020-04-20:
First I've copied the library locally, that works fine. But I would like it to easily be updated and distributed. So I tried working with the Platformio Library:

lib_deps = 
    lvgl@v6.1.2

According to the documentation I've added the lv_conf.h to the src folder and added these lines in the ini file:

build_flags =
  -D LV_LVG_H_INCLUDE_SIMPLE
  ; -I src Required to find lv_conf.h
  -I src

But I still get the error:
.pio\libdeps\nextGen\lvgl_ID6598\src\lv_core\lv_obj.h:19:30: fatal error: ../../../lv_conf.h: No such file or directory compilation terminated.

Any suggestions?

Thanks!
David

puzrin, 2020-04-20:
Does this project build with target `stm32f429_disco` works?

puzrin, 2020-07-03:
Close - outdated. All build tests pass https://travis-ci.org/lvgl/lv_platformio

closed 2020-07-03 by puzrin