lvgl/lv_platformio #19: Update to 7.0.0 Please!

sensei73, 2020-05-19:
First of all, thanks for this marvelous software and congratulation in your news achievements!

I'm so exited to test out the 7.0.0 version.
But it will be Wonderfull if you could update this simulator!

Thanks in advance.

Staars, 2020-07-26:
I did a short test with the benchmark-demo and it works for me.

You can check my fork, if you like:
https://github.com/Staars/lv_platformio

mbrdar, 2020-11-28:
@Staars thanks a lot it works! Whit lvgl 7.7.2 version I got some errors in printer demo but I just removed that demo end project is compiling and working perfectly!

bjwschaap, 2021-01-19:
I've updated this repo to the most recent LVGL version, and provided pull request #26

PR #26 (2021-01-20): Update demo for lvgl 7.9.1
I updated this repo to use the latest available platformio lvgl library version (7.9.1), and rewrote the demo to use the new v7 API.

Fixes #19
closed 2021-01-20 by puzrin