lvgl/lv_platformio #20: a new language

mostafajalal20, 2020-07-07:
hi,how can i add a new language such persian or arabic to this programm

kisvegabor, 2020-07-07:
To platformio or LVGL?

If LVGL see https://github.com/lvgl/lv_i18n
If PlatformIO you should ask in PlatformIOs forum instead: https://community.platformio.org/

closed 2020-07-07 by kisvegabor

mostafajalal20, 2020-07-07:
i should add persian or arabic language to LVGL without any writing problem such as sepearte letters from each other because in persian or arabic letter can Sticking together, and i haven't any idea for this.please guide me

kisvegabor, 2020-07-08:
Here you can learn about Persian support of LVGL: https://docs.lvgl.io/latest/en/html/overview/font.html#arabic-and-persian-support

The built-in `LV_FONT_DEJAVU_16_PERSIAN_HEBREW` might help to get started.

---

We use GitHub Issues for development-related discussions.

Please continue in the [Forum](https://forum.lvgl.io) to make the discussion visible for a larger community. You can simply log in with your GitHub account.