lvgl/lv_platformio #21: Something wrong in package manager

kisvegabor, 2020-07-08:
Hi,

I've just noticed that the library manager doesn't fetch lvgl correctly.
https://platformio.org/lib/show/6598/lvgl

Do you know why can it happen?

puzrin, 2020-07-08:
https://travis-ci.org/github/lvgl/lv_platformio/builds/706000370

Just re-executed build tests (you can do it too anytime). All pass.

What's exactly wrong & how to reproduce?

kisvegabor, 2020-07-08:
Versions look strange
[image]

All the downloads come from the last week.
[image]

So it seems something is messed up.

embeddedt, 2020-07-08:
It seems that it's not working with the automated release script (7.0.2 has some strange prefix on it).

kisvegabor, 2020-07-08:
I've noticed that `v` prefix was added back to the version string. 
I've removed it and updated the release script.

puzrin, 2020-07-08:
So, as far as i understand, question is not about package manager (local tool), but about registry & it's crawler.

Then i'd suggest ask at their forum. Also note, they planned to upgrade regisry and could break something https://community.platformio.org/t/embedded-template-library-v15-x-dissappeared-from-registry/14096/5?u=vit.

kisvegabor, 2020-07-08:
> So, as far as i understand, question is not about package manager (local tool), but about registry & it's crawler.

Ah, I see.

> Then i'd suggest ask at their forum. Also note, they planned to upgrade regisry and could break something

Let's give it a day, maybe the removed `v` prefix changes something.

puzrin, 2020-07-12:
I see 7.1.0 in registry

kisvegabor, 2020-07-13:
Yep, all look good after removing the `v` prefix.

Good to see that there are so many downloads from PIO too :) 

As the issue is solved I close it now. Thanks!

closed 2020-07-13 by kisvegabor