lvgl/lv_platformio #22: Devboard drivers cleanup

puzrin, 2020-07-12:
@kisvegabor Seems `/drivers/stm32f429_disco` duplicates a lot of files, already available in PIO. When board defined in config, that include appropriate HAL drivers + board-specific BSP drivers. Could you remove c/h pairs one-by-one an check if app build still works on hardware?

Candidates to remove - everything with ST copyright info.

kisvegabor, 2020-07-13:
I can take a look tomorrow.

kisvegabor, 2020-07-13:
I've checked it but found only the HAL driver in `.pio/build/stm32f429_disco/FrameworkHALDriver` but can't find the board specific files.

Do you know where are they?

puzrin, 2020-07-13:
See root point  `~/.platformio/packages/framework-stm32cube/`.

Then, inside `./f4/Drivers/BSP/STM32F429I-Discovery` and so on

kisvegabor, 2020-07-14:
Ah, thanks.

Still struggling with 
1. How to include files from framework. Only the absolute path worked so far.
2. How to add the BSP files to the build. HAL is built but e.g. `stm32f429i_discovery.c` doesn't.

puzrin, 2020-07-14:
Problem consists of several parts. Prior to start, please read https://community.platformio.org/t/how-in-include-disco-f429zi-headers/7001/13?u=vit

1. I removed copies of common BSP drivers https://github.com/lvgl/lv_platformio/commits/drop-files (in branch). Build is ok, but not tested on hardware
2. `stm32f429i_discovery.*` right now can not be included, because they still no fixed config (see topic above)
   - reported https://github.com/platformio/platform-ststm32/issues/413, let's postpone this
---

Pending:

- Seems `stm32f4_it.*` can be dropped (+ move SysTick_Handler to `driver.c`
- `system_stm32f4xx.c` - if not modified - drop it.

---

Also see https://github.com/platformio/platform-ststm32/blob/develop/builder/frameworks/stm32cube.py. All magic about directories happens here.

kisvegabor, 2020-07-16:
Thanks!

The `drop-files` branch workes fine. I removed `strm32f4_it.*` and `system_stm32f4xx.c`.

AFAIK `SystemClock_Config` is required. It seems the "blink" example runs with a default (very slow) clock.

My board has an 8MHz quartz so `HSE_VALUE` should be 8M. I don't know why was it 12M. (If `SystemClock_Config` thinks the quartz has a greater value it sets the PLL multiplier to a smaller number and the MCU will run slower.)

All these are updated in deaa864

puzrin, 2020-07-16:
1.

> My board has an 8MHz quartz so `HSE_VALUE` should be 8M.

This line [was added](https://github.com/lvgl/lv_platformio/commit/177aae6b0d39548e6263ced69c8a1df108feced0#diff-237285c047dad808f73698c139144e1aR43) by you. My quesion is, should it eхists there at all? It has defaults in hal headers.

Is it possible to just drop this line or something become broken?

2.

Do you have reasons to change [default env](https://github.com/lvgl/lv_platformio/commit/deaa86417fe1c6c8ef34cc616ebd754c30746fc7#diff-237285c047dad808f73698c139144e1aR12) or it was not intended?

3.

When complete, could you squash everything (including my changes) into single commit? Something like "stm: drivers cleanup"

kisvegabor, 2020-07-16:
> This line was added by you. 

I don't know why. Maybe just a typo.

> My quesion is, should it eхists there at all? It has defaults in hal headers.

The default value is 25M.

> Do you have reasons to change default env or it was not intended?

No intended. I'll change it back.

> When complete, could you squash everything (including my changes) into single commit? Something like "stm: drivers cleanup"

Sure, no problem.

puzrin, 2020-07-16:
All clean to me. Thanks for info.

1. https://github.com/lvgl/lv_platformio/blob/master/platformio.ini#L59 - please set to 8000000 and add comment above `# header's default is 25MHz, but board uses 8MHz crystal`
2. Squash everything in that branch to single commit.

I think this can be landed. Will create new issue when PIO will fix board headers include.

kisvegabor, 2020-07-17:
Done, merged to master.

closed 2020-07-17 by puzrin