lvgl/lv_platformio #23: Cleanup hal drivers (part 2)

puzrin, 2020-08-14:
Continue of https://github.com/lvgl/lv_platformio/issues/22.

Pio [fixed](https://github.com/platformio/platform-ststm32/issues/413) bug with search paths or our devboard and we can finish with the rest. I already removed some files https://github.com/lvgl/lv_platformio/commits/master.

@kisvegabor could you check content of ttf/touchpad src https://github.com/lvgl/lv_platformio/tree/master/hal/stm32f429_disco? Can those be reduced by reuse of board drivers from hal?

See `~/.platformio/packages/framework-stm32cube/f4/Drivers/BSP/STM32F429I-Discovery`

kisvegabor, 2020-08-18:
It seems the rest of the init function are not part of BSP. They are added manually in the official examples too. See: 
- [LTDC](https://github.com/STMicroelectronics/STM32CubeF4/blob/master/Projects/STM32F429I-Discovery/Examples/LTDC/LTDC_Display_2Layers/Src/main.c#L194)
- [DMA2D](https://github.com/STMicroelectronics/STM32CubeF4/blob/master/Projects/STM32F429I-Discovery/Examples/DMA2D/DMA2D_MemToMemWithBlending/Src/main.c#L125)
- [SDRAM](https://github.com/STMicroelectronics/STM32CubeF4/blob/master/Projects/STM32F429I-Discovery/Examples/FMC/FMC_SDRAM/Src/main.c#L265)

puzrin, 2020-08-18:
Ok. Let me ask different. There are 2 BSP files:

- ~/.platformio/packages/framework-stm32cube/f4/Drivers/BSP/STM32F429I-Discovery/stm32f429i_discovery_lcd.c
- ~/.platformio/packages/framework-stm32cube/f4/Drivers/BSP/STM32F429I-Discovery/stm32f429i_discovery_ts.c

Can those be reused by driver to reduce code size? A first glance, those already have everything and need only very thin wrapper.

kisvegabor, 2020-08-18:
Thanks!

I could drop most of the functions. Now tft.c is only 200 lines. I don't have the dev board with me now so I can test it only a little bit later.

kisvegabor, 2020-08-18:
I've tested an pushed the tft changes.

puzrin, 2020-08-19:
Thanks! I think now drivers looks "simple" and will no scare anyone :).

closed 2020-08-19 by puzrin

kisvegabor, 2020-08-20:
Looks great! Especially after removing the `tft` and `touchpad` directories.