lvgl/lv_platformio #25: Execute error on fresh install

gilphilbert, 2020-10-17:
I installed the pre-requisites and commented out the 32-bit line (I don't have the libs and 64-bit is just fine). Other than that, it's a clean install. Opened the folder in VSCode, ran `Task... Execute (Native)` and got the following error:
```
> Executing task: ./.pio/build/native/program <

/bin/bash: ./.pio/build/native/program: No such file or directory
The terminal process "/bin/bash '-c', './.pio/build/native/program'" failed to launch (exit code: 127).

Terminal will be reused by tasks, press any key to close it.
```

Any help appreciated!

puzrin, 2020-10-29:
They have broken compaibility in vscode 2.x extension. But you can make build and then run manually from terminal.

puzrin, 2020-10-29:
execue is defined here https://github.com/lvgl/lv_platformio/blob/master/.vscode/tasks.json

What OS do you use? Just checked - works in ubunu. What is in your `/.pio/build/native` folder?

Finkelrf, 2020-11-10:
I was getting the same error. The .pio/build/native folder was not been created, to fix I had to select env:native on "Switch PlatformIO Project Environment" on the bottom of the vccode window.

bjwschaap, 2020-11-23:
Thank you for sharing your solution @Finkelrf ! I had the same issue on OSX, switching to `env:native` solved the issue.

orzel, 2020-12-27:
Same problem, and the "fix" from @Finkelrf worked. Thanks :-)

puzrin, 2021-01-22:
This is caused by changes in pio v5+ & vsode plgin v2+

1. I've set default end to `native`
2. The rest moved to #30

closed 2021-01-22 by puzrin