lvgl/lv_platformio #27: `add-path` should be removed from CI

bjwschaap, 2021-01-21:
See: https://github.blog/changelog/2020-10-01-github-actions-deprecating-set-env-and-add-path-commands/

puzrin, 2021-01-21:
Note, i tried to do that, but got problems with sdl2 install/build.

Anyway, yes, this should be done somehow. In my projects i just dropped this part of build from CI, but that's not nice.

embeddedt, 2021-01-21:
I updated it to use the correct syntax [here](https://github.com/lvgl/lv_platformio/compare/27cfe50fbd926e6eaa3f4ca886f9aae21dfb45cf...33d3f5c9bef516d512ffa9c97666a011d0d87f04#diff-b803fcb7f17ed9235f1e5cb1fcd2f5d3b2838429d4368ae4c57ce4436577f03f).

@puzrin This should work for your project as well.

closed 2021-01-21 by embeddedt

puzrin, 2021-01-21:
@embeddedt you helped me very much! Thank you!