lvgl/lv_platformio #29: CI: force use more fresh compiler for `native` on OSX

puzrin, 2021-01-22:
**Problem:**

- `native` (emulator) build uses default OS's gcc compiler
- gcc in OSX is VERY old (they prefer llvm)

As a result, when you try to use some modern libs like ETL, build will fail. See https://github.com/puzrin/dispenser/blob/master/.github/workflows/ci.yml#L52-L54

---

I consider this as bug of PIO, but problem still exists.

andrew-elsayeh, 2022-01-19:
Yes, this is a good point! 

There is no gcc port for the Apple Silicone Macs (M1, M1 Pro and M1 Max)

puzrin, 2022-01-19:
Do PR, if you know how to fix https://github.com/lvgl/lv_platformio/blob/master/.github/workflows/ci.yml#L56-L76

bobwolff68, 2022-09-20:
After lots of gyrations, I did manage to compile and execute this project on a Mac M1. The change was simple. Add a few items to build_flags in platformio.ini
  -arch arm64
  -I /opt/homebrew/include
  -L /opt/homebrew/lib

fbiego, 2024-12-03:
Closing this issue as it’s resolved

closed 2024-12-03 by fbiego