lvgl/lv_platformio #32: is this a demo or sim? how to use the sim?

drunkscientist, 2021-02-26:
so im super confused and all of the docs apparently assume professional programmer, which im not. i was able to get it to load some random demo, but im trying to simulate my smartwatch forked from https://github.com/sharandac/My-TTGO-Watch. platformIO and vscode are already good, what else do i need? sim should be an extension for vscode or pIO or something...

puzrin, 2021-02-26:
This is demo how to use lvgl with pio, building multiple targets - bare metal & sim. But you are expected to know how to use pio.

closed 2021-02-26 by puzrin

drunkscientist, 2021-02-26:
thats not helpful... i know how to use pio, that doesnt tell me what to do with this, or how to simulate my project instead of the included demo

puzrin, 2021-02-26:
Sorry, but generic programming questions are out of this project scope.

drunkscientist, 2021-02-26:
how is "how to use THIS software" outside the scope of this software? this is not generic, this is "which files need to go where? to make the simulation of another thing that isnt the demo," which seems pretty specific to me. the readme reads like it assumes platformio is new, including instructions for installing vscode and pio... yay... then explains up to how to run the demo: "To build/execute, on PlafomIO tab expand desired env and click target:" great, still not trying to run the demo, trying to simulate my thing. the last thing listed in the readme is "Note, for emulator env upload also executes compiled binary." does that mean that i can use the compiled binary from wherever it gets compiled and drop it here to simulate it, or just that it has the same function as the execute button?

embeddedt, 2021-02-26:
I think there may be a misunderstanding here. LVGL simulators are evaluation/development environments for LVGL itself. They do not simulate specific hardware.

drunkscientist, 2021-02-26:
> I think there may be a misunderstanding here. 

this is definitely true. is this for the development **of** LVGL or for developing things **with** LVGL? all im trying to do, is simulate what will be displayed on my watch, without having to actually upload it to the watch.

embeddedt, 2021-02-26:
You can use this to develop things with LVGL, but it will not simulate your watch directly.

This is a generic simulator for designing LVGL user interfaces, but it does not include any specific APIs that you would find in your watch's firmware.

drunkscientist, 2021-02-27:
> This is a generic simulator for designing LVGL user interfaces, but it does not include any specific APIs that you would find in your watch's firmware.

ok but is there some part of this that i could include somewhere else that, when clicking execute, would simulate that page, probably with a custom build env in the target project? i dont expect it to be able to actually trigger the buttons for the applications/ watchface, im just trying to save myself the headache of having to make another update because im 5 pixels too big or buttons overlap etc. my apologies for being completely clueless still, i was linked to this project from [this page](https://docs.lvgl.io/v7/en/html/get-started/pc-simulator.html) and this was the first one i got working(and also the first i tried). maybe, can you tell me: does there exist something that will let me "preview" the output of an active project? this LOOKS like it could, but i admit im not an expert with pio yet and have no idea what would need to go where to make it do

embeddedt, 2021-02-27:
> does there exist something that will let me "preview" the output of an active project?

With embedded development, it is difficult to preview the output of a project easily, unless the project already has some type of emulation/simulation environment set up.

This is where the difference between a simulator and emulator really comes into play. Simulators only aim to replicate some of a device's behavior. In LVGL's case our simulator can only replicate the user interface. You can change the screen size, etc., but it's not a replacement for development on the actual target.

An emulator would replicate the hardware's behavior very closely to the point of being indistinguishable.

I think what you are really looking for is an emulator for your watch, but I'm not sure if such a thing exists. You would probably need to ask in that community.

There are two solutions which are commonly used to solve this problem:

* Structure your GUI code in such a way that it can be copied back and forth between the LVGL simulator project and the target project easily. If you set the screen size of the simulator to match the watch, you can get a feel for how things would look without having to download the code to the watch each time. The downside here is that your entire GUI has to be architected in such a way that it runs in two different environments.
* Make iterating on the real hardware faster, so that you can quickly deploy updates and see how things would look. For example, on my personal STM32 build environment, I can see the effects of a change within about 10-15 seconds.

drunkscientist, 2021-02-27:
mad thanks for this. the [documentation](https://docs.lvgl.io/v6/en/html/) lists key features as "Hardware independent to use with any microcontroller or display," i think i interpreted that to mean that gui code would be portable between environments, which it sounds like it can be, but isnt inherently...

embeddedt, 2021-02-28:
Correct. The LVGL APIs themselves are the same regardless of what environment you use them in, but if your GUI code is interacting with hardware or relying on a certain screen size, you have to mock those things yourself. By "hardware independent" we mean that the library itself doesn't depend on a specific display type or processor.