lvgl/lv_platformio #35: Update to v8 Please!

XIGUAjuice, 2021-06-11:
Does anyone work on v8 simulator?

kisvegabor, 2021-10-18:
It's great! Could you send a Pull request?

reflector-li, 2021-11-30:
> PS. after some code change I succefully run it with v.8.0.1 here is my code: https://github.com/smarq8/lv_platformio-simulator-v8.0.1 [image]

Could you help me solve it ? I download your version and compile it in VS code, but it still remind that There is no <SDL/>

> PS. after some code change I succefully run it with v.8.0.1 here is my code: https://github.com/smarq8/lv_platformio-simulator-v8.0.1 [image]

Could you help me to solve it ? I download your version and compile it in VS code, but it still remind that "<SDL2/SDL.h>" does not exist. The message is 
`> Executing task: C:\Users\12971\.platformio\penv\Scripts\platformio.exe run <

Processing emulator_64bits (platform: native@^1.1.3)
------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 2 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <lvgl> 8.0.1
|-- <lv_drivers> 8.0.1
|   |-- <lvgl> 8.0.1
Building in release mode
Compiling .pio\build\emulator_64bits\libfaf\lvgl\core\lv_disp.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\core\lv_event.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\core\lv_group.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\core\lv_indev.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\core\lv_indev_scroll.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\core\lv_obj.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\core\lv_obj_class.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\core\lv_obj_draw.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\core\lv_obj_pos.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\core\lv_obj_scroll.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\core\lv_obj_style.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\core\lv_obj_style_gen.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\core\lv_obj_tree.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\core\lv_refr.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\core\lv_theme.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\draw\lv_draw_arc.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\draw\lv_draw_blend.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\draw\lv_draw_img.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\draw\lv_draw_label.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\draw\lv_draw_line.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\draw\lv_draw_mask.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\draw\lv_draw_rect.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\draw\lv_draw_triangle.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\draw\lv_img_buf.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\draw\lv_img_cache.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\draw\lv_img_decoder.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\layouts\flex\lv_flex.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\layouts\grid\lv_grid.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\lv_extra.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\themes\basic\lv_theme_basic.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\themes\default\lv_theme_default.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\themes\mono\lv_theme_mono.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\widgets\animimg\lv_animimg.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\widgets\calendar\lv_calendar.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\widgets\calendar\lv_calendar_header_arrow.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\widgets\calendar\lv_calendar_header_dropdown.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\widgets\chart\lv_chart.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\widgets\colorwheel\lv_colorwheel.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\widgets\imgbtn\lv_imgbtn.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\widgets\keyboard\lv_keyboard.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\widgets\led\lv_led.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\widgets\list\lv_list.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\widgets\meter\lv_meter.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\widgets\msgbox\lv_msgbox.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\widgets\span\lv_span.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\widgets\spinbox\lv_spinbox.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\widgets\spinner\lv_spinner.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\widgets\tabview\lv_tabview.o
Compiling .pio\build\emulator_64bits\libfaf\lvgl\extra\widgets\tileview\lv_tileview.o
[... 107 lines cut ...]
compilation terminated.
*** [.pio\build\emulator_64bits\hal\sdl2\app_hal.o] Error 1
In file included from .pio\libdeps\emulator_64bits\lv_drivers\indev\keyboard.c:9:
.pio\libdeps\emulator_64bits\lv_drivers\indev\keyboard.h:33:34: fatal error: SDL2/SDL.h: No such file or directory
 #define MONITOR_SDL_INCLUDE_PATH <SDL2/SDL.h>
                                  ^
compilation terminated.
*** [.pio\build\emulator_64bits\lib08d\lv_drivers\indev\keyboard.o] Error 1
In file included from .pio\libdeps\emulator_64bits\lv_drivers\indev\mouse.c:9:
[... cut ...]
.pio\libdeps\emulator_64bits\lv_drivers\indev\mousewheel.h:33:34: fatal error: SDL2/SDL.h: No such file or directory
 #define MONITOR_SDL_INCLUDE_PATH <SDL2/SDL.h>
                                  ^
compilation terminated.
*** [.pio\build\emulator_64bits\lib08d\lv_drivers\indev\mousewheel.o] Error 1
============================================================= [FAILED] Took 9.08 seconds =============================================================

Environment      Status    Duration
---------------  --------  ------------
emulator_64bits  FAILED    00:00:09.082`

reflector-li, 2021-11-30:
> PS. after some code change I succefully run it with v.8.0.1 here is my code: https://github.com/smarq8/lv_platformio-simulator-v8.0.1 [image]

Could you help me solve it? I download the code of your version, but it still reminds that the <SDL2/SDL.h> does not exist. 
[image]

smarq8, 2021-11-30:
did you install msys2 and SLD library?

see instruction at readme and msys2 installation guide [here](https://www.msys2.org/)

```
Windows

Use MSYS2

pacman -S mingw-w64-x86_64-gcc mingw-w64-x86_64-SDL2

Add the path to your Mingw-w64 bin folder to the Windows PATH environment variable (usually C:\msys64\mingw64\bin). See instruction, 4.
```

reflector-li, 2021-12-03:
> did you install msys2 and SLD library?
>
> see instruction at readme and msys2 installation guide [here](https://www.msys2.org/)
>
> ```
> Windows
>
> Use MSYS2
>
> pacman -S mingw-w64-x86_64-gcc [...]

Thank you very much, I have installed them, and I had installed MinGW before. It may be confict with MSY32. Now I uninstalled the MinGW and only use the compiler in MSY32，and after restart the system, it works well.

escherstair, 2022-02-09:
The pull request has been closed without merging (it seems so)

richardclli, 2022-03-01:
I have made a pull request that works with LVGL v8.2

commit 338a1078 (2022-03-05): Merge pull request #41 from richardclli/lvgl-8.2

Updated sources for LVGL8.2, resolves #35
closed 2022-03-05 by puzrin