lvgl/lv_platformio #39: Can not find file:lvgl.h?

daihuangliang, 2021-12-26:
Can not find file:lvgl.h?

daihuangliang, 2021-12-26:
**************************************************************

   27 | #include "lvgl.h"
      |          ^~~~~~~~
In file included from .pio\libdeps\emulator_64bits\lv_drivers\display\monitor.c:9:
.pio\libdeps\emulator_64bits\lv_drivers\display\monitor.h:27:10: fatal error: lvgl.h: No such file or directory

**************************************************************
* Looking for lvgl.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:lvgl.h"
* Web  > https://platformio.org/lib/search?query=header:lvgl.h
*
**************************************************************

   27 | #include "lvgl.h"
      |          ^~~~~~~~
In file included from .pio\libdeps\emulator_64bits\lv_drivers\indev\mousewheel.c:9:
.pio\libdeps\emulator_64bits\lv_drivers\indev\mousewheel.h:27:10: fatal error: lvgl.h: No such file or directory

**************************************************************
* Looking for lvgl.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:lvgl.h"
* Web  > https://platformio.org/lib/search?query=header:lvgl.h
*
**************************************************************

   27 | #include "lvgl.h"
      |          ^~~~~~~~
In file included from .pio\libdeps\emulator_64bits\lv_drivers\indev\keyboard.c:9:
.pio\libdeps\emulator_64bits\lv_drivers\indev\keyboard.h:27:10: fatal error: lvgl.h: No such file or directory

**************************************************************
* Looking for lvgl.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:lvgl.h"
* Web  > https://platformio.org/lib/search?query=header:lvgl.h
*
**************************************************************

   27 | #include "lvgl.h"
      |          ^~~~~~~~
compilation terminated.
compilation terminated.
compilation terminated.
compilation terminated.
compilation terminated.
*** [.pio\build\emulator_64bits\lib0e6\lv_drivers\indev\mousewheel.o] Error 1
*** [.pio\build\emulator_64bits\lib0e6\lv_drivers\indev\keyboard.o] Error 1
*** [.pio\build\emulator_64bits\lib0e6\lv_drivers\indev\mouse.o] Error 1
*** [.pio\build\emulator_64bits\src\demo.o] Error 1
*** [.pio\build\emulator_64bits\lib0e6\lv_drivers\display\monitor.o] Error 1
src\main.c:12:10: fatal error: lvgl.h: No such file or directory

zoenggamming, 2021-12-30:
yes, I got the same problem.

ZK11111111, 2022-01-14:
> 兄弟问题解决了吗？

sonictl, 2022-01-15:
If you meet difficulties when getting started with lvgl on ESP32, next three links may help:

[[ESP32_lvgl]Tips for getting started with lvgl project on ESP32](https://sonictl.github.io/jekyll/2022/01/15/p20220115140254.html)

[[ESP32_lvgl]Start a lvgl project on ESP32 with vsCode, platformIO on Arduino platform
](https://sonictl.github.io/jekyll/2022/01/15/p20220115140154.html)

[[ESP32_lvgl]Complile the lvgl sample proj. in visual studio Code (vsCode) with help of ESP-IDF extension](https://sonictl.github.io/jekyll/2022/01/11/p20220111210954.html)

commit 485d0bd7 (2022-01-15): Fix build
- close #39
closed 2022-01-15 by puzrin