lvgl/lv_platformio #44: error: 'for' loop initial declaration used outside C99 mode

MarcoRizz, 2022-04-28:
Hi, I'm trying to start with lvgl.
Following the README steps for Windows, at the execution in PlatformIO the output is the error

```
.pio\libdeps\emulator_64bits\lvgl\src\core\lv_obj.c: In function 'lv_obj_event':
.pio\libdeps\emulator_64bits\lvgl\src\core\lv_obj.c:734: error: 'for' loop initial declaration used outside C99 mode
*** [.pio\build\emulator_64bits\lib0e3\lvgl\core\lv_obj.o] Error 1
.pio\libdeps\emulator_64bits\lvgl\src\core\lv_refr.c: In function 'draw_buf_rotate_90':
.pio\libdeps\emulator_64bits\lvgl\src\core\lv_refr.c:825: error: 'for' loop initial declaration used outside C99 mode
.pio\libdeps\emulator_64bits\lvgl\src\core\lv_refr.c:829: error: 'for' loop initial declaration used outside C99 mode
.pio\libdeps\emulator_64bits\lvgl\src\core\lv_refr.c: In function 'draw_buf_rotate_90_sqr':
.pio\libdeps\emulator_64bits\lvgl\src\core\lv_refr.c:858: error: 'for' loop initial declaration used outside C99 mode
.pio\libdeps\emulator_64bits\lvgl\src\core\lv_refr.c:859: error: 'for' loop initial declaration used outside C99 mode
*** [.pio\build\emulator_64bits\lib0e3\lvgl\core\lv_refr.o] Error 1
```

Do I have to change some setting of PlatformIO?
Thanks

puzrin, 2022-04-28:
No ideas. At least, CI pass https://github.com/lvgl/lv_platformio/actions/workflows/ci.yml

May be you have old compiler in your win? What env do you use?

MarcoRizz, 2022-04-29:
In VScode terminal at command `g++ --version` the output is
```
g++ --version
g++.exe (GCC) 4.2.1-sjlj (mingw32-2)
Copyright (C) 2007 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO 
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
even if MSYS2 is updated. I'm trying to fix it.

puzrin, 2022-04-29:
Try to reproduce steps from CI config:

https://github.com/lvgl/lv_platformio/blob/master/.github/workflows/ci.yml#L80-L105

Probably, missed to sync something in readme. I have no windows to experiment with.