lvgl/lv_platformio #51: Support lvgl released V8.3.0?

Pingye007, 2023-02-12:
Hi, I am new to lvgl/Platform IO and I use this project to have a try.It is very successful(8.2.0).
But I want to update lvgl to released 8.3.0 and continue to reuse this project again.Compiling was fine but executing was broken.I don't where goes wrong and no issues related to this similiar situation.
I want to figure it out but I'm not a professional developer.
This picture shows compile with 8.3.0 is ok.
[image]

Picture under is the broken captured.`platformio.ini` as well.
[image]

Pingye007, 2023-02-12:
``` plantxt
; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; http://docs.platformio.org/page/projectconf.html

[platformio]
default_envs = emulator_64bits

; Shared options
[env]
build_flags =
  ; Don't use lv_conf.h. Tweak params via platfom.ini.
  -D LV_CONF_SKIP
  -D LV_CONF_INCLUDE_SIMPLE
  ; Enable LVGL demo, remove when working on your own project
  -D LV_USE_DEMO_WIDGETS=1
  ; Add more defines below to overide lvgl:/src/lv_conf_simple.h
lib_deps =
  ; Use direct URL, because package registry is unstable
  ;lvgl@~7.11.0
  lvgl=https://github.com/lvgl/lvgl/archive/refs/tags/v8.3.0.zip
lib_archive = false

[env:emulator_64bits]
platform = native@^1.1.3
extra_scripts = support/sdl2_build_extra.py
build_flags =
  ${env.build_flags}
  ; -D LV_LOG_LEVEL=LV_LOG_LEVEL_INFO
  -D LV_LOG_PRINTF=1
  ; Add recursive dirs for hal headers search
  !python -c "import os; print(' '.join(['-I {}'.format(i[0].replace('\x5C','/')) for i in os.walk('hal/sdl2')]))"
  -lSDL2
  ; SDL drivers options
  -D LV_LVGL_H_INCLUDE_SIMPLE
  -D LV_DRV_NO_CONF
  -D USE_SDL
  -D SDL_HOR_RES=480
  -D SDL_VER_RES=320  
  -D SDL_ZOOM=1
  -D SDL_INCLUDE_PATH="\"SDL2/SDL.h\""

  ; LVGL memory options, setup for the demo to run properly
  -D LV_MEM_CUSTOM=1
  -D LV_MEM_SIZE="(128U * 1024U)"
  
lib_deps =
  ${env.lib_deps}
  ; Use direct URL, because package registry is unstable
  ;lv_drivers@~7.9.0
  lv_drivers=https://github.com/lvgl/lv_drivers/archive/refs/tags/v8.3.0.zip
build_src_filter =
  +<*>
  +<../hal/sdl2>
  ; Force compile LVGL demo, remove when working on your own project
  +<../.pio/libdeps/emulator_64bits/lvgl/demos>

[env:emulator_32bits]
extends = env:emulator_64bits
build_flags =
  ${env:emulator_64bits.build_flags}
  -m32
build_src_filter =
  +<*>
  +<../hal/sdl2>
  +<../.pio/libdeps/emulator_32bits/lvgl/demos>

[env:stm32f429_disco]
platform = ststm32@^8.0.0
board = disco_f429zi
framework = stm32cube
build_flags =
  ${env.build_flags}
  -D LV_LOG_LEVEL=LV_LOG_LEVEL_NONE
  ; header's default is 25MHz, but board uses 8MHz crystal
  -D HSE_VALUE=8000000
  ; Add recursive dirs for hal headers search
  !python -c "import os; print(' '.join(['-I {}'.format(i[0].replace('\x5C','/')) for i in os.walk('hal/stm32f429_disco')]))"
lib_deps =
  ${env.lib_deps}
  BSP-ili9341
  BSP-stmpe811
build_src_filter =
  +<*>
  +<../hal/stm32f429_disco>
  ; Force compile LVGL demo, remove when working on your own project
  +<../.pio/libdeps/stm32f429_disco/lvgl/demos>

```

Pingye007, 2023-02-12:
@puzrin 
I'll be very appreciated if you can give me some ideas about this problem!

Pingye007, 2023-02-12:
Oh!!
I feel extremely sorry about my silly thoughts! t_t
It is the version uncompatible between lvgl and its driver.
I should been aware of this. Driver still is 8.2.0 and lvgl should be 8.3.0.

mars3142, 2023-08-10:
> Driver still is 8.2.0 and lvgl should be 8.3.0.

On my macOS, I can update both to 8.3.0 and it's running without any issues.

fbiego, 2024-12-20:
Closing this issue, feel free to reopen if you have problems with v9

closed 2024-12-20 by fbiego