lvgl/lv_platformio #52: Win11_PlatfomIO_Build_Issues

TigoTas, 2023-03-02:
Hi,

First I have to say that the work being done with LVGL is amazing. I hope to contribute to it.

I'm having issues with the porting to PlatformIO, though.

Followed instalation as described on the main page. Also as described here [https://code.visualstudio.com/docs/cpp/config-mingw#_prerequisites](url).

I checked #7 and #17 for possible fixes. But build fails

Any help is appreciated.

I get the following on the terminal:

Executing task: C:\Users\tigo_\.platformio\penv\Scripts\platformio.exe run 

Processing emulator_64bits (platform: native@^1.1.3)
------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 3 compatible libraries
Scanning dependencies...
Dependency Graph
|-- lvgl @ 8.2.0
|-- lv_drivers @ 8.2.0
|   |-- lvgl @ 8.2.0
Building in release mode
Compiling .pio\build\emulator_64bits\libcbf\lv_drivers\sdl\sdl.o
<command-line>: fatal error: SDL2/SDL.h: No such file or directory
compilation terminated.
Compiling .pio\build\emulator_64bits\.pio\libdeps\emulator_64bits\lvgl\demos\stress\assets\lv_font_montserrat_28_compr_az.o
Compiling .pio\build\emulator_64bits\.pio\libdeps\emulator_64bits\lvgl\demos\stress\lv_demo_stress.o
Compiling .pio\build\emulator_64bits\.pio\libdeps\emulator_64bits\lvgl\demos\widgets\assets\img_clothes.o
Compiling .pio\build\emulator_64bits\.pio\libdeps\emulator_64bits\lvgl\demos\widgets\assets\img_demo_widgets_avatar.o
Compiling .pio\build\emulator_64bits\.pio\libdeps\emulator_64bits\lvgl\demos\widgets\assets\img_lvgl_logo.o
Compiling .pio\build\emulator_64bits\.pio\libdeps\emulator_64bits\lvgl\demos\widgets\lv_demo_widgets.o
Compiling .pio\build\emulator_64bits\hal\sdl2\app_hal.o
*** [.pio\build\emulator_64bits\libcbf\lv_drivers\sdl\sdl.o] Error 1
Compiling .pio\build\emulator_64bits\src\main.o
hal\sdl2\app_hal.c:3:10: fatal error: SDL2/SDL.h: No such file or directory
    3 | #include <SDL2/SDL.h>
      |          ^~~~~~~~~~~~
compilation terminated.
*** [.pio\build\emulator_64bits\hal\sdl2\app_hal.o] Error 1
============================================== [FAILED] Took 9.67 seconds ==============================================

Environment      Status    Duration
---------------  --------  ------------
emulator_64bits  FAILED    00:00:09.675
======================================== 1 failed, 0 succeeded in 00:00:09.675 ======================================== 

 *  The terminal process "C:\Users\tigo_\.platformio\penv\Scripts\platformio.exe 'run'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

TigoTas, 2023-03-02:
Reinstalled everything and got it working.

closed 2023-03-02 by TigoTas

jiahuan-chen, 2023-05-24:
thanks fool cat，i successd it