lvgl/lv_platformio #53: .pio/libdeps/emulator_64bits/lv_drivers/sdl/sdl.c:53:10: fatal error: 'SDL2/SDL.h' file not found

one-lucky-pig, 2023-03-03:
`
Compiling .pio/build/emulator_64bits/liba79/lv_drivers/display/drm.o
Compiling .pio/build/emulator_64bits/liba79/lv_drivers/display/fbdev.o
Compiling .pio/build/emulator_64bits/liba79/lv_drivers/gtkdrv/gtkdrv.o
Compiling .pio/build/emulator_64bits/liba79/lv_drivers/indev/AD_touch.o
Compiling .pio/build/emulator_64bits/liba79/lv_drivers/indev/FT5406EE8.o
Compiling .pio/build/emulator_64bits/liba79/lv_drivers/indev/XPT2046.o
Compiling .pio/build/emulator_64bits/liba79/lv_drivers/indev/evdev.o
Compiling .pio/build/emulator_64bits/liba79/lv_drivers/indev/libinput.o
Compiling .pio/build/emulator_64bits/liba79/lv_drivers/indev/xkb.o
Compiling .pio/build/emulator_64bits/liba79/lv_drivers/sdl/sdl.o
Compiling .pio/build/emulator_64bits/liba79/lv_drivers/sdl/sdl_gpu.o
Compiling .pio/build/emulator_64bits/liba79/lv_drivers/wayland/wayland.o
Compiling .pio/build/emulator_64bits/liba79/lv_drivers/win32drv/win32drv.o
Compiling .pio/build/emulator_64bits/liba79/lv_drivers/win_drv.o
Compiling .pio/build/emulator_64bits/.pio/libdeps/emulator_64bits/lvgl/demos/benchmark/assets/img_benchmark_cogwheel_alpha16.o
Compiling .pio/build/emulator_64bits/.pio/libdeps/emulator_64bits/lvgl/demos/benchmark/assets/img_benchmark_cogwheel_argb.o
.pio/libdeps/emulator_64bits/lv_drivers/sdl/sdl.c:53:10: fatal error: 'SDL2/SDL.h' file not found
#include SDL_INCLUDE_PATH
         ^~~~~~~~~~~~~~~~
<command line>:12:26: note: expanded from here
#define SDL_INCLUDE_PATH "SDL2/SDL.h"
                         ^~~~~~~~~~~~
1 error generated.
Compiling .pio/build/emulator_64bits/.pio/libdeps/emulator_64bits/lvgl/demos/benchmark/assets/img_benchmark_cogwheel_chroma_keyed.o
Compiling .pio/build/emulator_64bits/.pio/libdeps/emulator_64bits/lvgl/demos/benchmark/assets/img_benchmark_cogwheel_indexed16.o
Compiling .pio/build/emulator_64bits/.pio/libdeps/emulator_64bits/lvgl/demos/benchmark/assets/img_benchmark_cogwheel_rgb.o
Compiling .pio/build/emulator_64bits/.pio/libdeps/emulator_64bits/lvgl/demos/benchmark/assets/lv_font_bechmark_montserrat_12_compr_az.c.o
*** [.pio/build/emulator_64bits/liba79/lv_drivers/sdl/sdl.o] Error 1
Compiling .pio/build/emulator_64bits/.pio/libdeps/emulator_64bits/lvgl/demos/benchmark/assets/lv_font_bechmark_montserrat_16_compr_az.c.o
=============================================== [FAILED] Took 2.28 seconds ===============================================

Environment      Status    Duration
---------------  --------  ------------
emulator_64bits  FAILED    00:00:02.280
========================================== 1 failed, 0 succeeded in 00:00:02.280 ==========================================

 *  The terminal process "platformio 'run', '--environment', 'emulator_64bits'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
`

mars3142, 2023-03-22:
I'm running on macOS and do have the same issue. I installed SDL2 via brew.

laytll, 2023-04-12:
I added this to fix that error under build_flags in platformio.ini:

  -I /opt/homebrew/Cellar/sdl2/2.24.2/include
  -L /opt/homebrew/Cellar/sdl2/2.24.2/lib

you might need to check your version in the path

ohmytime, 2023-04-29:
> I added this to fix that error under build_flags in platformio.ini:
> 
> -I /opt/homebrew/Cellar/sdl2/2.24.2/include -L /opt/homebrew/Cellar/sdl2/2.24.2/lib
> 
> you might need to check your version in the path

Thanks, it works ~
[image]

bobtista, 2023-07-21:
Note replace the version number with your own, eg today it's 2.28.1 for me

sqrtroot, 2023-11-05:
A way to do this independent of version numbers is the by adding the following lines:

```
  !find /opt/homebrew/Cellar/sdl2 -name "SDL.h" | xargs dirname | sed "s/^/-I /"
  !find /opt/homebrew/Cellar/sdl2 -name "libSDL2.a" | xargs dirname | sed "s/^/-L /"
```

so the full build_flags looks like this:
```ini
build_flags =
  ${env.build_flags}
  ; -D LV_LOG_LEVEL=LV_LOG_LEVEL_INFO
  -D LV_LOG_PRINTF=1
  ; Add recursive dirs for hal headers search
  !python3 -c "import os; print(' '.join(['-I {}'.format(i[0].replace('\x5C','/')) for i in os.walk('hal/sdl2')]))"
  -lSDL2
  ; SDL drivers options
  -D LV_LVGL_H_INCLUDE_SIMPLE
  -D LV_DRV_NO_CONF
  -D USE_SDL
  -D SDL_HOR_RES=480
  -D SDL_VER_RES=320  
  -D SDL_ZOOM=1
  -D SDL_INCLUDE_PATH="\"SDL2/SDL.h\""

  ; LVGL memory options, setup for the demo to run properly
  -D LV_MEM_CUSTOM=1
  -D LV_MEM_SIZE="(128U * 1024U)"
  
  ; SDL2 includes
  !find /opt/homebrew/Cellar/sdl2 -name "SDL.h" | xargs dirname | sed "s/^/-I /"
  !find /opt/homebrew/Cellar/sdl2 -name "libSDL2.a" | xargs dirname | sed "s/^/-L /"
```

pkerspe, 2024-01-12:
@kisvegabor 
would be helpful to update the documentation with the fix provided above, which also worked fine for me on MacOS!

kisvegabor, 2024-01-14:
Hi,

Could you send a Pull request?

kisvegabor, 2024-01-14:
Merged, thank you. I'm closing this issue.

closed 2024-01-14 by kisvegabor