lvgl/lv_platformio #54: Simulator: Printf not working at runtime | Simulator starts twice

MasterTim17, 2023-04-29:
Hello,
im using the simulator with platformio and used this repo example.
When i upload the project the simulator starts. When i close it another windows opens. This is essential for the following problem

I want to debug the code with printf command. So i defined following in platform.ini file:
```
-D LV_USE_LOG=1
-D LV_LOG_PRINTF=1
```
Default log level is : `-D LV_LOG_LEVEL=LV_LOG_LEVEL_WARN`
During runtime i dont get an output. When i close the Simulator window, the print output is now seen on the output. The second windows open, but the print output from it is not seen after closing.

However, when i change the log level to `-D LV_LOG_LEVEL=LV_LOG_LEVEL_INFO` the print output can now be seen during runtime. thi s isnt useful, because too much output is beeing listed.

I hope the problem is understandable. Is this a bug or is there a solution?

3254547566, 2023-07-21:
I am also facing the same problem. Is there a solution now

Yutu0k, 2024-09-25:
Still waiting for a solution...

kisvegabor, 2024-10-03:
Hi,

How do you print the logs? With `lv_log()` you need to append a  `"\n"`. E.g. `lv_log("Hello %d world\n", 5)`. `LV_LOG_USER("Hello %d world", 5)` append the `\n` internally.

Recently `fflush` was added to printf logging. See [here](https://github.com/lvgl/lvgl/blob/master/src/misc/lv_log.c#L104). Does it solve the problem?

kisvegabor, 2024-10-14:
Does it mean the problem is solved? :slightly_smiling_face: