lvgl/lv_platformio #58: Why do x86_64 files appear when compiling in native environment on MAC Silicon

CNSTJiatongLee, 2024-01-27:
I set platform=native on Mac M1 and it compiles using my native gcc compiler. Why is my native gcc compiler in ARM64 format and able to compile x86_64 files? This caused an error when linking to sdl2 in arm64 format that homebrew downloaded
[image]
[image]
[image]

tylerwowen, 2024-03-01:
I ran into the same problem. 

For my situation, the Platform IO installed was Darwin_x86_64. I don't know why it happened. My guess is that when I first installed the PlatformIO IDE extension, I was in a VSCode running in Intel mode, which could be a result of starting the VSCode from an Intel mode shell. 
```
pio  system info
--------------------------  ---------------------------------------------
PlatformIO Core             6.1.13
Python                      3.11.7-final.0
System Type                 darwin_x86_64
```

To fix, I did the following
1. Uninstall the PlatformIO ID extension from VSCode.
2. Remove the `$HOME/.platformio` created by the PlatformIO IDE
3. Reinstall the PlatformIO. Now I got the right version. 

```
pio  system info
--------------------------  ---------------------------------------------
PlatformIO Core             6.1.13
Python                      3.9.6-final.0
System Type                 darwin_arm64
```

navanchauhan, 2024-03-06:
Just in case someone else stumbles upon this issue. I gave up on using the VSCode IDE and instead just installed paltformio via homebrew 

```
brew install platformio
```

I also had to change `python` to `python3` in the `platformio.ini` file (Or, you can set an alias from python -> python)

fbiego, 2024-12-03:
Closing this issue as it’s resolved

closed 2024-12-03 by fbiego