lvgl/lv_platformio #59: Support lvgl released V9.0.0?

BlynkGO, 2024-01-29:
I have tried to use lvgl v9.0.0, however it occurs some errors as the following

[image]

How to solve them?
Thank you.

kisvegabor, 2024-01-29:
Hi,

You can see how the API changed in v9 [here](https://docs.lvgl.io/master/CHANGELOG.html).

However in v9 there is a built in SDL driver that you can use very easily. See [here](https://github.com/lvgl/lv_port_pc_eclipse/blob/master/main.c#L111) and enable [here](https://github.com/lvgl/lv_port_pc_eclipse/blob/master/lv_conf.h#L832-L839).

fbiego, 2024-12-20:
Closing this issue, feel free to reopen if you still have problems with v9

closed 2024-12-20 by fbiego