lvgl/lv_platformio #60: fatal error: SDL2/SDL.h: No such file or directory

Joaquin500, 2024-01-31:
Executing task: C:\Users\Joaquin\.platformio\penv\Scripts\platformio.exe run --target execute --environment emulator_64bits 

Processing emulator_64bits (platform: native@^1.1.3)
--------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 2 compatible libraries
Scanning dependencies...
Dependency Graph
|-- lvgl @ 8.2.0
|-- lv_drivers @ 8.2.0
Building in release mode
Compiling .pio\build\emulator_64bits\lib58c\lv_drivers\sdl\sdl.o
<command-line>: fatal error: SDL2/SDL.h: No such file or directory
compilation terminated.

Any help would be more than welcome, as I'm a newbie with all these tools.

kisvegabor, 2024-02-01:
Hi,

Have you seen installed SDL?

kisvegabor, 2024-02-01:
This guide can be helpful to install SDL on Windiws: https://lazyfoo.net/tutorials/SDL/01_hello_SDL/windows/index.php

kisvegabor, 2024-02-02:
> C:/msys64/ucrt64/bin/../lib/gcc/x86_64-w64-mingw32/13.1.0/../../../../x86_64-w64-mingw32/bin/ld.exe: cannot find -lSDL2: No such file or directory

It indicates that SDL2 is not installed or not seen by PlatformIO.

Joaquin500, 2024-02-02:
I have it installed in 2 different places and both paths added to the PATH enviroment variable as indicated. Any way, I have given up and cleaned it all. Once again, thank you for your advice :-)

closed 2024-02-02 by Joaquin500

kisvegabor, 2024-02-05:
I'm sorry that we couldn't figure it out :slightly_frowning_face:

lgrosbois-AtlasLaser, 2024-04-25:
No problem.
My post was mainly to store somewhere the resolution of the issue I had.
First developer reflex is to google the error, so finding a hint to resolution is always a good sign :)

Have a good day 👍