lvgl/lv_platformio #61: Missing guideline / Difficulties to use library with platformIO

MainWire, 2024-03-15:
Hi,

After spending some time and eventually managing to install the LV_PlatformIO library, I am encountering a few build errors I cannot fix.
The main problem seems to be that I cannot get a correct main.cpp file content, and I am having difficulties understanding the platformIO instructions

## My installation :

- lvgl @ 8.2.0
- The project folder I am using (lv_platformio master project) [Github](https://github.com/lvgl/lv_platformio)
-  framework-espidf @ 3.50102.240122 (5.1.2)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-esp32ulp @ 1.23500.220830 (2.35.0)
 - toolchain-xtensa-esp32 @ 12.2.0+20230208
 - PlatformIO.ini file : [Platformio.ini](https://dpaste.org/mjwRd)
 - Cmake file :
  ```
  cmake_minimum_required(VERSION 3.16.0)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(LVGL_C_Main)
```

## Attempt n°1 (demo_create was not declared)

1. Using the instructions and the _main.c_ available at the following link [Platformio support](https://github.com/lvgl/lv_port_esp32?tab=readme-ov-file#platformio-support) 
2. and after correcting the line `#include "demo.h"` to `#include "demos/lv_demos.h"` because of the `demo.h` not found

I am getting the following build error : `src/main.cpp:28:5: error: 'demo_create' was not declared in this scope; did you mean 'lv_obj_create'?`

## Attempt n°2 (monitor.h no such file or directory)

1. Using the following main.cpp [main.cpp dpaste link](https://dpaste.org/wjZ4L)
2. After Moving the '_hal_' folder to _lib_ folder, because vscode wasn't finding the file in main.cpp
3. And Correcting the path to hal with the following : `#include "..\lib\hal\sdl2\app_hal.c"` (otherwise app_hal isn't found)
4. Downloading _SDL2-devel-2.30.1-VC.zip _ at the following link [SDL download](https://github.com/libsdl-org/SDL/releases) and putting it in the lib folder (with the hal folder)

I am getting the following error : `src/..\lib\hal\sdl2\app_hal.c:4:10: fatal error: display/monitor.h: No such file or directory`

I am unsure of how I should proceed to use this library with platformIO

I would be really grateful to anyone providing some help !

MainWire, 2024-03-20:
I would be really happy for any help I may have 👍  I still can't figure out how to make the library compile with the examples sketch. Ask anything you need !

fbiego, 2024-11-27:
Have you tried the latest update with LVGL 9?

ghisbo, 2024-12-08:
I fully agree, no way to get somewhere with these guidelines.

fbiego, 2024-12-09:
Can you share what challenge you are having, we can find a way to improve the guidelines

ghisbo, 2024-12-09:
I followed the procedure outlined here. I can produce the emulator but i do not find an env for the "Elecrow3.5 with TFT touch". based on ESP32S3. I can program the ESP32S3 but i cannot get anything to display. I have been looking in Google but no working result.
What would help is an example with the most basic feature ( label) complete with platformio definition ( in my case the Elecrow 3.5 TFT touch (parallel)

ghisbo, 2024-12-10:
Sounds very good