lvgl/lv_platformio #62: Unable to run the example because SDL2 cannot be linked on macOS with the M1 Max chip

e7h4n, 2024-03-20:
## What I encountered

I got an error message when I chose "Project Tasks / emulator_64bits / General / Execute":

```
Linking .pio/build/emulator_64bits/program
ld: warning: ignoring file '/opt/homebrew/Cellar/sdl2/2.30.1/lib/libSDL2-2.0.0.dylib': found architecture 'arm64', required architecture 'x86_64'
ld: Undefined symbols:
  _SDL_CreateRenderer, referenced from:
      _window_create in sdl.o
...
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

## What I did

* Install SDL2 in macOS with `brew install sdl2`
* git clone this repo
* uncomment SDL2 includes of `[env:emulator_64bits]` in platformio.ini
```
!find /opt/homebrew/Cellar/sdl2 -name "include" | sed "s/^/-I /"
!find /opt/homebrew/Cellar/sdl2 -name "libSDL2.a" | xargs dirname | sed "s/^/-L /"
```
* Open this project in vscode with platform.io extension
* Chose "Project Tasks / emulator_64bits / General / Execute" in vscode platform.io extension

## What I Expected

Compile & Linking should be successes, and emulator GUI should appear in my screen.

## What I tried

* I google with some keywords: `lvgl emulator m chip macos` and found a post of this: [https://community.platformio.org/t/lvgl-simulator-on-macos-ventura/31714/10], it looks like the same error. But this post don't have any progress.

## My Environment

* macOS 14.2.1 (23C71)
* Macbook Pro 2021
* CPU: M1 Max
* RAM: 64 GB
* VSCode: 1.87.2 (Universal)
* PlatformIO: 3.3.3
* SDL2
```
brew info sdl2
==> sdl2: stable 2.30.1 (bottled), HEAD
Low-level access to audio, keyboard, mouse, joystick, and graphics
https://www.libsdl.org/
/opt/homebrew/Cellar/sdl2/2.30.1 (93 files, 6.5MB) *
  Poured from bottle using the formulae.brew.sh API on 2024-03-20 at 12:01:36
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/s/sdl2.rb
License: Zlib
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 60,852 (30 days), 147,383 (90 days), 599,804 (365 days)
install-on-request: 10,521 (30 days), 25,143 (90 days), 103,846 (365 days)
build-error: 363 (30 days)
```

## Project Dependency

```

Resolving emulator_64bits dependencies...
Platform native @ 1.2.1 (required: native @ ^1.1.3)
├── contrib-piohome @ 3.4.4 (required: platformio/contrib-piohome)
├── framework-arduino-sam @ 1.6.12 (required: platformio/framework-arduino-sam)
├── framework-arduinoespressif32 @ 3.20014.231204 (required: platformio/framework-arduinoespressif32)
[... 453 lines cut ...]
      _window_update in sdl.o
  _SDL_RenderCopy, referenced from:
      _window_update in sdl.o
  _SDL_RenderPresent, referenced from:
      _window_update in sdl.o
  _SDL_SetEventFilter, referenced from:
      _sdl_init in sdl.o
  _SDL_SetTextureBlendMode, referenced from:
      _window_create in sdl.o
  _SDL_StartTextInput, referenced from:
      _sdl_init in sdl.o
  _SDL_UpdateTexture, referenced from:
      _window_update in sdl.o
clang: error: linker command failed with exit code 1 (use -v to see invocation)
*** [.pio/build/emulator_64bits/program] Error 1
============================================================================================================ [FAILED] Took 23.58 seconds ============================================================================================================

Environment      Status    Duration
---------------  --------  ------------
emulator_64bits  FAILED    00:00:23.584
======================================================================================================= 1 failed, 0 succeeded in 00:00:23.584 =======================================================================================================

 *  The terminal process "platformio 'run', '--target', 'execute', '--environment', 'emulator_64bits'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 
```

e7h4n, 2024-03-20:
I tried to install 2.26.4 SDL2, which someone said works in m-chip mbp, but I still got this error. Maybe I should build an x86_64 arch manually.

e7h4n, 2024-03-20:
Another try:

* I built SDL2 from source with "cmake -DCMAKE_OSX_ARCHITECTURES=x86_64 .." to x86_64 arch
* Link this library from platform.io with
```
  !find ../SDL/build -name "include" | sed "s/^/-I /"
  !find ../SDL/build -name "libSDL2.a" | xargs dirname | sed "s/^/-L /"
```

But still got other error messages:

```
...
  _CFStringGetMaximumSizeForEncoding, referenced from:
      _COREAUDIO_GetDefaultAudioInfo in libSDL2.a[142](SDL_coreaudio.m.o)
      _build_device_list in libSDL2.a[142](SDL_coreaudio.m.o)
  _CFStringGetSystemEncoding, referenced from:
      _HIDGetDeviceProduct in libSDL2.a[162](SDL_syshaptic.c.o)
  _CFStringGetTypeID, referenced from:
      _get_string_property_utf8 in libSDL2.a[36](SDL_hidapi.c.o)
      _get_string_property_utf8 in libSDL2.a[36](SDL_hidapi.c.o)
      _get_string_property in libSDL2.a[36](SDL_hidapi.c.o)
      _get_string_property in libSDL2.a[36](SDL_hidapi.c.o)
  _CFURLCreateWithBytes, referenced from:
      _SDL_SYS_OpenURL in libSDL2.a[141](SDL_sysurl.m.o)
  _CFUUIDGetConstantUUIDWithBytes, referenced from:
      _DARWIN_JoystickInitRumble in libSDL2.a[160](SDL_iokitjoystick.c.o)
      _SDL_SYS_HapticEffectType in libSDL2.a[162](SDL_syshaptic.c.o)
      _SDL_SYS_HapticEffectType in libSDL2.a[162](SDL_syshaptic.c.o)
      _SDL_SYS_HapticEffectType in libSDL2.a[162](SDL_syshaptic.c.o)
      _SDL_SYS_HapticEffectType in libSDL2.a[162](SDL_syshaptic.c.o)
      _SDL_SYS_HapticEffectType in libSDL2.a[162](SDL_syshaptic.c.o)
      _SDL_SYS_HapticEffectType in libSDL2.a[162](SDL_syshaptic.c.o)
      _SDL_SYS_HapticEffectType in libSDL2.a[162](SDL_syshaptic.c.o)
      _SDL_SYS_HapticEffectType in libSDL2.a[162](SDL_syshaptic.c.o)
      _SDL_SYS_HapticEffectType in libSDL2.a[162](SDL_syshaptic.c.o)
      _SDL_SYS_HapticEffectType in libSDL2.a[162](SDL_syshaptic.c.o)
      _SDL_SYS_HapticEffectType in libSDL2.a[162](SDL_syshaptic.c.o)
      ...
  _CGAcquireDisplayFadeReservation, referenced from:
      _Cocoa_SetDisplayMode in libSDL2.a[172](SDL_cocoamodes.m.o)
  _CGAssociateMouseAndMouseCursorPosition, referenced from:
      _Cocoa_WarpMouseGlobal in libSDL2.a[173](SDL_cocoamouse.m.o)
      _Cocoa_SetRelativeMouseMode in libSDL2.a[173](SDL_cocoamouse.m.o)
      _Cocoa_SetRelativeMouseMode in libSDL2.a[173](SDL_cocoamouse.m.o)
      -[Cocoa_WindowListener mouseMoved:] in libSDL2.a[179](SDL_cocoawindow.m.o)
  _CGCaptureAllDisplays, referenced from:
      _Cocoa_SetDisplayMode in libSDL2.a[172](SDL_cocoamodes.m.o)
  _CGColorGetConstantColor, referenced from:
      -[SDLView drawRect:] in libSDL2.a[179](SDL_cocoawindow.m.o)
      -[SDLView updateLayer] in libSDL2.a[179](SDL_cocoawindow.m.o)
  _CGDisplayBounds, referenced from:
      _Cocoa_GetDisplayBounds in libSDL2.a[172](SDL_cocoamodes.m.o)
  _CGDisplayCapture, referenced from:
      _Cocoa_SetDisplayMode in libSDL2.a[172](SDL_cocoamodes.m.o)
  _CGDisplayCopyAllDisplayModes, referenced from:
      _Cocoa_GetDisplayDPI in libSDL2.a[172](SDL_cocoamodes.m.o)
      _Cocoa_GetDisplayModes in libSDL2.a[172](SDL_cocoamodes.m.o)
  _CGDisplayCopyDisplayMode, referenced from:
      _Cocoa_InitModes in libSDL2.a[172](SDL_cocoamodes.m.o)
      _Cocoa_GetDisplayModes in libSDL2.a[172](SDL_cocoamodes.m.o)
  _CGDisplayFade, referenced from:
      _Cocoa_SetDisplayMode in libSDL2.a[172](SDL_cocoamodes.m.o)
      _Cocoa_SetDisplayMode in libSDL2.a[172](SDL_cocoamodes.m.o)
      _Cocoa_SetDisplayMode in libSDL2.a[172](SDL_cocoamodes.m.o)
  _CGDisplayIDToOpenGLDisplayMask, referenced from:
      _Cocoa_GL_CreateContext in libSDL2.a[174](SDL_cocoaopengl.m.o)
  _CGDisplayIOServicePort, referenced from:
      _Cocoa_GetDisplayName in libSDL2.a[172](SDL_cocoamodes.m.o)
  _CGDisplayIsMain, referenced from:
[... 943 lines cut ...]
      -[Cocoa_WindowListener listen:] in libSDL2.a[179](SDL_cocoawindow.m.o)
  _objc_sync_enter, referenced from:
      -[SDLOpenGLContext setWindow:] in libSDL2.a[174](SDL_cocoaopengl.m.o)
      -[SDLOpenGLContext setWindow:] in libSDL2.a[174](SDL_cocoaopengl.m.o)
      _ScheduleContextUpdates in libSDL2.a[179](SDL_cocoawindow.m.o)
  _objc_sync_exit, referenced from:
      -[SDLOpenGLContext setWindow:] in libSDL2.a[174](SDL_cocoaopengl.m.o)
      -[SDLOpenGLContext setWindow:] in libSDL2.a[174](SDL_cocoaopengl.m.o)
      -[SDLOpenGLContext setWindow:] in libSDL2.a[174](SDL_cocoaopengl.m.o)
      -[SDLOpenGLContext setWindow:] in libSDL2.a[174](SDL_cocoaopengl.m.o)
      _ScheduleContextUpdates in libSDL2.a[179](SDL_cocoawindow.m.o)
      _ScheduleContextUpdates in libSDL2.a[179](SDL_cocoawindow.m.o)
  _objc_unsafeClaimAutoreleasedReturnValue, referenced from:
      _Cocoa_RegisterApp in libSDL2.a[168](SDL_cocoaevents.m.o)
      _CreateApplicationMenus in libSDL2.a[168](SDL_cocoaevents.m.o)
      _CreateApplicationMenus in libSDL2.a[168](SDL_cocoaevents.m.o)
      _CreateApplicationMenus in libSDL2.a[168](SDL_cocoaevents.m.o)
      _CreateApplicationMenus in libSDL2.a[168](SDL_cocoaevents.m.o)
      _CreateApplicationMenus in libSDL2.a[168](SDL_cocoaevents.m.o)
      _CreateApplicationMenus in libSDL2.a[168](SDL_cocoaevents.m.o)
      _CreateApplicationMenus in libSDL2.a[168](SDL_cocoaevents.m.o)
      _CreateApplicationMenus in libSDL2.a[168](SDL_cocoaevents.m.o)
      ...
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```

e7h4n, 2024-03-20:
I did it!
My solution is add `-arch arm64` to build_flags, like

```
[env:emulator_64bits]
platform = native@^1.1.3
extra_scripts = support/sdl2_build_extra.py
build_flags =
  ${env.build_flags}
  -arch arm64
```

Then everythings works like a charm 😄

e7h4n, 2024-03-20:
This exception is related to Visual Studio Code because everything is OK when I execute the PlatformIO task in the shell instead of running "execute task" in VSCode.

I guess the platform extension in VSCode installs an x86_64-based Python environment, which leads Scons to think the building environment is x86_64. So, if I execute the `pio run` command in the shell by a natively installed Python environment, Scons could get the correct architecture.

fbiego, 2024-11-28:
Seems the issue is now resolved
Also related to issues #58 #29

fbiego, 2024-12-03:
Closing this issue as it’s resolved

closed 2024-12-03 by fbiego