lvgl/lv_platformio #65: CAN NOT WORK ON WIN10!

linkua, 2024-05-16:
Dependency Graph
|-- lvgl @ 9.1.0
Building in release mode
=====================================
Compiling .pio\build\emulator_64bits\libf53\lvgl\drivers\sdl\lv_sdl_keyboard.o
Compiling .pio\build\emulator_64bits\libf53\lvgl\drivers\sdl\lv_sdl_mouse.o
Compiling .pio\build\emulator_64bits\libf53\lvgl\drivers\sdl\lv_sdl_mousewheel.o
Compiling .pio\build\emulator_64bits\libf53\lvgl\drivers\sdl\lv_sdl_window.o
<command-line>: fatal error: SDL2/SDL.h: No such file or directory
compilation terminated.
*** [.pio\build\emulator_64bits\libf53\lvgl\drivers\sdl\lv_sdl_keyboard.o] Error 1
<command-line>: fatal error: SDL2/SDL.h: No such file or directory
compilation terminated.
*** [.pio\build\emulator_64bits\libf53\lvgl\drivers\sdl\lv_sdl_mouse.o] Error 1
<command-line>: fatal error: SDL2/SDL.h: No such file or directory
compilation terminated.
*** [.pio\build\emulator_64bits\libf53\lvgl\drivers\sdl\lv_sdl_mousewheel.o] Error 1
<command-line>: fatal error: SDL2/SDL.h: No such file or directory
compilation terminated.
*** [.pio\build\emulator_64bits\libf53\lvgl\drivers\sdl\lv_sdl_window.o] Error 1

kisvegabor, 2024-05-20:
Have you followed the Windows guide [here](https://github.com/lvgl/lv_platformio?tab=readme-ov-file#install-sdl-drivers)?

tylercamp, 2024-06-03:
I'm also having the same issue - I've installed MSYS2, ran the mentioned commands, and added `C:\msys64\mingw64\bin` to both System and User `PATH`. I can see `SDL2.dll` in that folder, and can see the `SDL2` folder in the adjacent `include` folder

---

Just after typing that I checked for any relevant config files in the repo folder on my machine, and in `.vscode/launch.json` I saw references to a `%USERPROFILE%/mingw` folder which I apparently prepared ~a year ago

After removing that from my `PATH` (and restarting VS Code) I was able to run the demo

kisvegabor, 2024-06-05:
Happy that finally it's working.

> After removing that from my PATH (and restarting VS Code) I was able to run the demo

@linkua can you confirm it?

closed 2024-06-21 by linkua

brightproject, 2025-06-09:
> After removing that from my `PATH` (and restarting VS Code) I was able to run the demo

@kisvegabor and @tylercamp 

Unfortunately, there is an unclear and confusing line at the end of the instructions.
I don't understand why `MSYS2` should be launched in `Win10` using `pacman`?
On `Win10`, the build occurs with an error, but under `Win10` with `WSL1`, the build is performed without errors.

> https://github.com/lvgl/lv_platformio/issues/75

It seems to me that the [instructions](https://github.com/lvgl/lv_platformio?tab=readme-ov-file#install-visual-studio-code) for `Win10` need to be updated or removed altogether, they don’t work.

tylercamp, 2025-06-12:
The note about `pacman` is directly from Microsoft's guidance on Windows https://code.visualstudio.com/docs/cpp/config-mingw#_prerequisites

Note that this is meant to be run within the msys2 CLI, I can find it by searching for `msys` in the start menu

brightproject, 2025-06-12:
> mingw

https://github.com/lvgl/lv_platformio/issues/65#issuecomment-2966570466

Thank you? already fine🙂

> https://github.com/lvgl/lv_platformio/issues/75#issuecomment-2959945777