lvgl/lv_platformio #72: simple tutorial on how to get this running on different boards

jappyjan, 2025-01-07:
i am trying to get this running with my board (https://www.waveshare.com/wiki/ESP32-C6-LCD-1.47) but it just wont work...
i allways end up with (different) compilation errors...

can you please provide simple instrucitons on how to adapt the given example to other boards?

jappyjan, 2025-01-07:
simply replacing the board with the esp32-c6-devkitc-1 results in the following compile error:
```
In file included from hal/esp32/app_hal.cpp:8:
hal/esp32/displays/lgfx_elecrow_3_5.hpp:23:11: error: 'Bus_Parallel16' in namespace 'lgfx' does not name a type
   23 |     lgfx::Bus_Parallel16 _bus_instance;
      |           ^~~~~~~~~~~~~~
hal/esp32/displays/lgfx_elecrow_3_5.hpp: In constructor 'LGFX::LGFX()':
hal/esp32/displays/lgfx_elecrow_3_5.hpp:33:24: error: '_bus_instance' was not declared in this scope; did you mean '_touch_instance'?
   33 |             auto cfg = _bus_instance.config();
      |                        ^~~~~~~~~~~~~
      |                        _touch_instance
```

dev-web54, 2025-01-22:
> i am trying to get this running with my board (https://www.waveshare.com/wiki/ESP32-C6-LCD-1.47) but it just wont work... i allways end up with (different) compilation errors...
> 
> can you please provide simple instrucitons on how to adapt the given example to other boards?

Did you find the solution? I have the same problem

kisvegabor, 2025-02-06:
cc @fbiego

fbiego, 2025-02-06:
@jappyjan @dev-web54 The waveshare board uses a different configuration (SPI). We are currently using [LovyanGFX library](https://github.com/lovyan03/LovyanGFX) for the drivers. You would need a matching configuration for your display. To get you started, please take a look at the steps below.
NB: You might need to adjust the parameters as I can't test on my end

First, create a new file in the displays folder eg `LGFX_WAVESHARE_C6_1_47.hpp` and copy/paste the configuration below

```
#pragma once
#define LGFX_USE_V1
#include <LovyanGFX.hpp>

/**
 * Recommended board settings for platformio.ini file
 * 
 * board = esp32-c6-devkitc-1
 * 
 */
#define WIDTH 172
#define HEIGHT 320

class LGFX : public lgfx::LGFX_Device
{

    lgfx::Panel_ST7789 _panel_instance;
    lgfx::Bus_SPI _bus_instance;
    lgfx::Light_PWM _light_instance;

public:
    LGFX(void)
    {
        {
            auto cfg = _bus_instance.config();

            // SPI bus settings
            cfg.spi_host = SPI2_HOST; // Select the SPI to use ESP32-S2,C3 : SPI2_HOST or SPI3_HOST / ESP32 : VSPI_HOST or HSPI_HOST
            // * Due to the ESP-IDF version upgrade, VSPI_HOST, The HSPI_HOST specification is deprecated, so if you get an error, use SPI2_HOST or SPI3_HOST instead.
            cfg.spi_mode = 0;                       // Set SPI communication mode (0 ~ 3)
            cfg.freq_write = 80000000;              // SPI time (up to 80MHz, four or five inputs divided by 80MHz to get an integer)
            cfg.freq_read = 20000000;               // SPI time when connected cfg.spi_3wire = true; // Set true if receiving is done via MOSI pin
            cfg.use_lock = true;                    // Usage lock time setting true
            cfg.dma_channel = SPI_DMA_CH_AUTO;      // Set the DMA channel to use (0=DMA not used / 1=1ch / 2=ch / SPI_DMA_CH_AUTO=automatic setting) // * Due to the ESP-IDF version upgrade, SPI_DMA_CH_AUTO (automatic setting) is now recommended for the DMA channel. Specifying 1ch or 2ch is no longer recommended.
            cfg.pin_sclk = 7;                    // Set the SPI SCLK pin number
            cfg.pin_mosi = 6;                    // Set the SPI CLK pin number
            cfg.pin_miso = -1;                    // Set the SPI MISO pin number (-1 = disable)
            cfg.pin_dc = 15;                        // Set the SPI D/C pin number (-1 = disable)
            _bus_instance.config(cfg);              // Reflect the setting value to the bus.
            _panel_instance.setBus(&_bus_instance); // Set the bus to the panel.
        }

        {                                        // Set display panel control.
            auto cfg = _panel_instance.config(); // Get the structure for display panel settings.

            cfg.pin_cs = 14;   // Pin number to which CS is connected (-1 = disable)
            cfg.pin_rst = 21;  // pin number where RST is connected (-1 = disable)
            cfg.pin_busy = -1; // pin number to which BUSY is connected (-1 = disable)

            // * The following setting values ​​are set to general default values ​​for each panel, and the pin number (-1 = disable) to which BUSY is connected, so please try commenting out any unknown items.

            cfg.memory_width = WIDTH;  // Maximum width supported by driver IC
            cfg.memory_height = HEIGHT; // Maximum height supported by driver IC
            cfg.panel_width = WIDTH;   // actual displayable width
            cfg.panel_height = HEIGHT;  // actual displayable height
            cfg.offset_x = 0;        // Panel offset in X direction
            cfg.offset_y = 0;        // Panel offset in Y directioncfg.offset_rotation = 2;
            cfg.dummy_read_pixel = 8;
            cfg.dummy_read_bits = 1;
            cfg.readable = false;
            cfg.invert = false;
            cfg.rgb_order = false;
            cfg.dlen_16bit = true;
            cfg.bus_shared = true;

            _panel_instance.config(cfg);
        }
        {                                        // Set backlight control. (delete if not necessary)
            auto cfg = _light_instance.config(); // Get the structure for backlight configuration.

            cfg.pin_bl = 22;     // pin number to which the backlight is connected
            cfg.invert = false;  // true to invert backlight brightness
            cfg.freq = 44100;    // backlight PWM frequency
            cfg.pwm_channel = 0; // PWM channel number to use

            _light_instance.config(cfg);
            _panel_instance.setLight(&_light_instance); // Sets the backlight to the panel.
        }

        setPanel(&_panel_instance); // Sets the panel to use.
    }
};

LGFX tft;

```

Then add it to `app_hal.cpp`
```
/* include only one display settings */
// #include "displays/lgfx_wt32sc01_plus.hpp"
// #include "displays/lgfx_elecrow_3_5.hpp"
#include "displays/LGFX_WAVESHARE_C6_1_47.hpp" /* <-- add  */
```