lvgl/lv_platformio #73: Unable to mount Win11 file system

ShermanWann, 2025-03-06:
I want to test an PNG image located in the Win11 path using the demo, so I made the following changes.

[image]

[image]

[image]

There is no error during build, but when I upload to the emulator_64bits, it reports that the file cannot be found. I'm sure I have placed the file in the correct directory.

[image]

BTW, I would like to mention that the original version without any modifications runs PERFECTLY.

kisvegabor, 2025-03-25:
Hi,

A few notes:
1. Do not modify `lv_conf_inernal.h` as it's internal. Set the path instead in the `ini` file
2. You should reference the file like this `lv_image_set_src(img, "D:/battery_charging.png")`. See [here](https://docs.lvgl.io/master/details/main-modules/fs.html).

kisvegabor, 2025-03-25:
cc @fbiego

fbiego, 2025-03-26:
Currently investigating. I'm also not able to load an image

```
  LV_LOG_WARN("FS ready %d", lv_fs_is_ready('A'));
  /* Confirm if file exists */
  if (!file_exists("C:/Users/felix/Documents/lvgl-images/AC.png"))
  {
    LV_LOG_WARN("File not found or cannot be opened");
  } else {
    LV_LOG_WARN("File accessible");
  }

  lv_obj_t *img = lv_image_create(lv_screen_active());
  lv_image_set_src(img, "A:C:/Users/felix/Documents/lvgl-images/AC.png");
  lv_obj_center(img);
```
```
[Warn]  (0.178, +0)      main: FS ready 1 main.cpp:75
[Warn]  (0.179, +1)      main: File accessible main.cpp:82
[Error] (0.179, +0)      image_decoder_get_info: File open failed: 12 lv_image_decoder.c:332
[Warn]  (0.179, +0)      lv_image_set_src: failed to get image info: A:C:/Users/felix/Documents/lvgl-images/AC.png lv_image.c:189
```