lvgl/lv_platformio #75: Attempts to run the simulator in WIN10 on WSL1

brightproject, 2025-06-09:
When I compile the code, everything goes great.

[image]

>  *  Executing task in folder lv_platformio: platformio run 
> 
> Processing emulator_64bits (platform: native@^1.1.3)
> -------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
> Verbose mode can be enabled via `-v, --verbose` option
> LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ soft
> Found 1 compatible libraries
> Scanning dependencies...
> Dependency Graph
> |-- lvgl @ 9.1.0
> Building in release mode
> `.pio/build/emulator_64bits/program' is up to date.
> ============================= [SUCCESS] Took 3.73 seconds =============================
> 
> Environment      Status    Duration
> ---------------  --------  ------------
> emulator_64bits  SUCCESS   00:00:03.734
> ============================= 1 succeeded in 00:00:03.734 =============================
>  *  Terminal will be reused by tasks, press any key to close it. 

When I follow the instructions `build/execute`

> https://github.com/lvgl/lv_platformio?tab=readme-ov-file#buildrun

 I get an error

> *  Executing task in folder lv_platformio: platformio run --target upload --environment emulator_64bits 
> 
> Processing emulator_64bits (platform: native@^1.1.3)
> ----------------------------------------------------------------------------------------------------------------------------------------------------
> Verbose mode can be enabled via `-v, --verbose` option
> LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ soft
> Found 1 compatible libraries
> Scanning dependencies...
> Dependency Graph
> |-- lvgl @ 9.1.0
> Building in release mode
> Configuring upload protocol...
> Executing .pio/build/emulator_64bits/program
> error: XDG_RUNTIME_DIR not set in the environment.
> error: XDG_RUNTIME_DIR not set in the environment.

Previously I tried to run

> https://github.com/lvgl/lv_platformio

In Windows on vscode, and there, when building the project, I got the following errors:

> Compiling .pio\build\emulator_64bits\libe9a\lvgl\drivers\sdl\lv_sdl_keyboard.o
> Compiling .pio\build\emulator_64bits\libe9a\lvgl\drivers\sdl\lv_sdl_mouse.o
> <command-line>: fatal error: SDL2/SDL.h: No such file or directory
> compilation terminated.
> *** [.pio\build\emulator_64bits\libe9a\lvgl\drivers\sdl\lv_sdl_keyboard.o] Error 1
> <command-line>: fatal error: SDL2/SDL.h: No such file or directory
> compilation terminated.
> *** [.pio\build\emulator_64bits\libe9a\lvgl\drivers\sdl\lv_sdl_mouse.o] Error 1
> ============================== [FAILED] Took 11.69 seconds ==============================
> 
> Environment      Status    Duration
> ---------------  --------  ------------
> emulator_64bits  FAILED    00:00:11.690
> ============================== 1 failed, 0 succeeded in 00:00:11.690 ==============================

 Here it is clearly visible that the compiler cannot find the path to `SDL2`, although I know for sure that `SDL2` is installed on the `PC` with `Windows`, because I successfully run `SDL2` in another simulator

> https://github.com/lvgl/lv_port_pc_vscode

[image]

I would be glad if someone knows the solution to the problems and tells me how to solve them.

It's just that another simulator is created for projects under `lvgl 9.3` and does not allow running codes for `Arduino`.
P.S. In general, all my attempts to run my examples run into one or another problem, it's already becoming uninteresting and soon the enthusiasm will disappear, everything is too confusing and crooked.

fbiego, 2025-06-10:
Hello @brightproject 
Can you try below;
replace
https://github.com/lvgl/lv_platformio/blob/7b6b7b2d0cceb449292dde1fe69e37d97d70d86a/platformio.ini#L48
with the SDL relative path
```
; -L C:/msys64/mingw64/lib/ ; also try with this uncommented
-D SDL_INCLUDE_PATH="\"C:/msys64/mingw64/include/SDL2/SDL.h\""
```

and also replace
https://github.com/lvgl/lv_platformio/blob/7b6b7b2d0cceb449292dde1fe69e37d97d70d86a/hal/sdl2/app_hal.c#L3
with 
```
#include SDL_INCLUDE_PATH
```

brightproject, 2025-06-10:
Hello @fbiego 🙂
Thank you for your help, but unfortunately I still couldn't run `lv_platformio` in the Linux subsystem on Windows `WSL`.
`Vscode` itself runs great using a command in the bash

> code .

In `vscode` I compile firmware for `Arduino`, including the `lv_platformio`, which I already wrote about above.
But unfortunately, apparently my `wsl` does not see the folders:

> C:/msys64/mingw64/include/SDL2/SDL.h

I also tried to put the whole folder in the `wsl`

> -D SDL_INCLUDE_PATH="\".home/user/Documents/Projects/lv_platformio/msys64/mingw64/include/SDL2/SDL.h\""

But this did not help, so I stayed with the working 

> vscode + windows 10 + lv_platformio + mingw64 + sdl2

I made a short video, basically duplicating the manual here for `Windows 10:

> https://github.com/lvgl/lv_platformio?tab=readme-ov-file#install-sdl-drivers

But the manual does not indicate that installation via pacman should be done after installing and running `mingw64`.
It also doesn't show how to add the path to `mingw64` in the Windows PATH environment variable.

Unfortunately, I also forgot to include this in the video, so I'll write it in text:.

> 1. Press WIN + R
> 2. Enter the command sysdm.cpl and go to the ADDITIONAL tab
> 3. Click on ENVIRONMENT VARIABLES
> 4. Click on Path in the upper window, where the user environment variables for Admin
> 5. In the window that appears, add to the free bottom line
> C:\msys64\mingw64\bin and click OK and close.
> After that, restart the `vscode` and everything as in the video - EXECUTE and a window with the LVGS simulator will appear.
> Thank you for your attention.

https://www.youtube.com/watch?v=AZl9oFWWZUs

brightproject, 2025-06-14:
Hello @fbiego 🙂
Is there any way to run this code examples in `lvgl 8.3.11`?

I was planning to run some of my projects for version `lvgl 8.3.11` in the simulator, but I ran into a problem with the lack of `sdl2` drivers for version `lvgl 8.3.11`.

Comparing the version repositories, it becomes clear that for `lvgl 8.3.11` there apparently were no `sdl2` or drivers yet.

> https://github.com/lvgl/lvgl/tree/release/v8.3/src

and

> https://github.com/lvgl/lvgl/tree/release/v9.3/src

At the same time, the project

> https://github.com/m5stack/lv_m5_emulator/

Allows you to run version `8.3` and `9.3` by simply switching the settings in `platformio.ini`

brightproject, 2025-06-14:
> in v8 the sdl drivers were not included in the library, check the repo here before it was updated to v9
> 
> https://github.com/lvgl/lv_platformio/tree/b68b2d8e230840b7a10a16cd591ffede02f422a4

Thank you.
Is this a repository for version `8.3`?
For some reason I didn't find any releases by version.

> https://github.com/lvgl/lv_platformio/releases

brightproject, 2025-06-23:
> Hello [@brightproject](https://github.com/brightproject)

Good day @fbiego 

Could you tell me please, how to correctly add my code to the main file of the `lv_platformio` simulator to animate and animate some of the objects I drew?

In the case of ready-made examples, this is done by simply adding a function, for example:

```
int main(void)
{
	lv_init();

	hal_setup();

        lv_demo_music();

	hal_loop();
}
```

in the main loop of the main file and the project works and everything moves.

In the classic application and the code that I write to load onto a real hardware device, I have an Arduino environment for building the code, its format is well-known:

```
#include <Arduino.h>

void setup() {

}

void loop() {

}
```

It is present in the main project file

> https://github.com/lvgl/lv_platformio/blob/7b6b7b2d0cceb449292dde1fe69e37d97d70d86a/src/main.cpp#L17

and it can even be connected by uncommenting the preprocessor directive

> #define TEST

But I couldn't do it, because when compiling code not for real hardware, you need a library

> #include <Arduino.h>

In general, I'm interested in running my code in a loop

```
int main(void)
{

}
```

Let's say, as I wrote earlier, for real hardware I write code like this:

```
#include "Arduino.h"
#include <ui.h>

#define TEST

#ifdef TEST

const int min_speed = 10; // Minimum value for speed
const int max_speed = 100; // Maximum value for speed
const int increment_speed = 1; // Increment step
int direction_speed = 1; // Direction of speed change

#endif

void setSpeed(int speed, int *a, int *b) {

  // Set values ​​with a difference of ±1 from the set speed
  *a = speed - 1; // speedFrom
  *b = speed + 1; // speedTo

}

static void PFD_Widget_TaskInit(void)
{

  // static void PFD_Widget_TaskInit(void *pvParameters)
  setSpeed(speed, &speedFrom, &speedTo);
  lv_arc_set_bg_angles(ui_SpeedARC, speedToAngleArc(speedFrom), speedToAngleArc(speedTo));

}

void setup()
{

  ui_init();

} // SETUP

void loop()
{

  #ifdef TEST

  speed += increment_speed * direction_speed;

  if (speed >= max_speed) {

    direction_speed = -1; // Change to decrease

  } else if (speed <= min_speed) {

    direction_speed = 1; // Change to increase

  }

  PFD_Widget_TaskInit();

} // LOOP
```

That is, earlier my code in a loop incremented and decremented the value of some variable, in this case speed, and substituted it into the value of the object drawing function, which was responsible for drawing the speed arrow

> lv_arc_set_bg_angles

Now I don’t understand how to use the simulator, I tried by analogy to place my code from the setup section into the `hal_setup() ` section, and from the loop section into the `hal_loop()` section;
Adding the path to my `UI` to the `app_hal.c` file

> #include <ui.h>

To sum up what I have written, at the moment I can only run examples from the `LVGL` library, animated with moving graphics and very beautiful, and my examples of `UI` without movement.

This is roughly how it looks on real hardware

https://github.com/user-attachments/assets/4c9062ff-0bcc-4900-91a1-59aad1c0d298

But I really want to check the work of my `UI` in motion with animations in the `LVGL` simulator🙂

brightproject, 2025-06-26:
Yes, thank you for your response🙂
I understand the difference between the Arduino structures and the classic scheme with
```
int main(int argc, char **argv)
{
  while(1) {

  }
}
```
My question was how to set your own logic for the application in the simulator so that it all works in an infinite loop.
After all, I need to put the code, for example, an increment of some value in

> hal_loop();

I do this and do not get the desired result.
The structure of the simulator is very confusing and unclear.
Now I can only run examples from the demo and some third-party projects in it that use animation based on `LVGL` functions, such as

> lv_timer_create

Which work from the `SETUP ` section.

fbiego, 2025-06-26:
You can also add code to the loop 
https://github.com/lvgl/lv_platformio/blob/7b6b7b2d0cceb449292dde1fe69e37d97d70d86a/hal/sdl2/app_hal.c#L51-L57

In this project, I use main.cpp as the entry point, then lvgl, ui, and other things are handled separately. The UI folder is shared
https://github.com/fbiego/windows-10-mobile-lvgl/blob/main/src/main.cpp