lvgl/lv_platformio #76: Use `${PROJECT_LIBDEPS_DIR}` instead of hard coded `../.pio/libdeps` in platformio.ini

atoomnetmarc, 2025-06-21:
I cannot compile the project because this fails:

```ini
  ; Force compile LVGL demo, remove when working on your own project
  +<../.pio/libdeps/emulator_64bits/lvgl/demos>
```

This hard codes the path of the .pio/libdeps directory. 

I have configured my .pio to a different directory. 

Could it be an idea to use https://docs.platformio.org/en/latest/projectconf/interpolation.html?

The resulting piece in platformio.ini:

```ini
  ; Force compile LVGL demo, remove when working on your own project
  +<${PROJECT_LIBDEPS_DIR}/emulator_64bits/lvgl/demos>
```

```bash
pio run --target envdump|grep -a2 PROJECT_LIBDEPS_DIR
  'LIBS': ['SDL2'],
  'LIBSOURCE_DIRS': [ '/home/marc/Downloads/lv_platformio/lib',
                      '$PROJECT_LIBDEPS_DIR/$PIOENV',
                      '/home/marc/.platformio/lib'],
  'LIBSUFFIX': '.a',
--
  'PROJECT_DIR': '/home/marc/Downloads/lv_platformio',
  'PROJECT_INCLUDE_DIR': '/home/marc/Downloads/lv_platformio/include',
  'PROJECT_LIBDEPS_DIR': '/tmp/.pio/lv_platformio-63e142e34b/libdeps',
  'PROJECT_PACKAGES_DIR': '/home/marc/.platformio/packages',
  'PROJECT_SRC_DIR': '/home/marc/Downloads/lv_platformio/src',
--
  'SRC_FILTER': [ '+<*>',
                  '+<../hal/sdl2>',
                  '+<${PROJECT_LIBDEPS_DIR}/emulator_64bits/lvgl/demos>'],
  'STATIC_AND_SHARED_OBJECTS_ARE_THE_SAME': 0,
  'TARGET_ARCH': None,
```

brightproject, 2025-06-23:
> I cannot compile the project

You must set the settings in `platformio.ini`

```
; Shared options
[env]
build_flags =
  -D LV_USE_DEMO_WIDGETS=1

build_src_filter =
  +<../.pio/libdeps/emulator_64bits/lvgl/demos>
```

You added a lot of garbage to `platformio.ini`

atoomnetmarc, 2025-06-23:
I guess you must be correct.

closed 2025-06-23 by atoomnetmarc